verilog-to-routing/vtr-verilog-to-routing #105: Invalid ODIN II output on MCML benchmark

kmurray, 2015-10-02:
This causes a failure in ABC on the MCML benchmark.  The ABC output is as follows:
```
The 4-input library started with 43906 nodes and 24772 subgraphs. Time =   0.04 sec
Line 4194: Cube "1-1-" has size different from the fanin count (2).
Reading network from file has failed.
````

It appears the BLIF produced by ODIN is invalid. The offending structure in mcml.odin.blif is:
```
.names top^LOGICAL_EQUAL~86260^LOGICAL_AND~449815 top^MULTI_PORT_MUX~86259^LOGICAL_NOT~86261 top^MULTI_PORT_MUX~86259^MUX_2~265159
1-1- 1
-1-1 1
````
where the number of inputs/outputs does not match the truth table dimensions.

This failure first occurred in the April 26th, 2015 weekly regression (on SVN revision 4600), and last passed in the March 15th, 2015 (on SVN revision 4593).

This makes the SVN commit 4594 the most likely suspect:
```
commit 00aef7d29fd8e01d8ed53b988101007bee5b7e00
Author: Bipin Kumar <bbipinkumarb@gmail.com>
Date:   Tue Mar 17 19:17:24 2015 +0000

    changing the name of driver signal pins in case of hard blocks for null signal names
    
    
    git-svn-id: https://vtr-verilog-to-routing.googlecode.com/svn/trunk@4594 8e3573b8-cf2c-4f14-ef6d-137439e28b8b

 ODIN_II/SRC/output_blif.c | 45 +++++++++++++++++++++++++++------------------
 1 file changed, 27 insertions(+), 18 deletions(-)
```

To reproduce the issue, from the root of the VTR source tree run:
```
./vtr_flow/scripts/run_vtr_task.pl regression_tests/vtr_reg_weekly/vtr_reg_qor/
```

To save time, I recommend editing `vtr_flow/tasks/regression_tests/vtr_reg_weekly/vtr_reg_qor/config/config.txt` to only run the MCML benchmark.

bipinkumarb, 2015-10-02:
Working on the issue.

kmurray, 2016-01-22:
Fixed in 6efd301a051723b09199673a7483e5f6c0057161

closed 2016-01-22 by kmurray