verilog-to-routing/vtr-verilog-to-routing #106: cannot build vtr

berbagci, 2015-10-05:
I'm getting the following error when I do "make". What am I missing?
Thanks

"In file included from SRC/route/route_tree_timing.h:4,
                 from SRC/draw/intra_logic_block.h:14,
                 from SRC/draw/draw.c:30:
SRC/route/connection_based_routing.h:115: error: expected nested-name-specifier before ‘CBRR’
SRC/route/connection_based_routing.h:115: error: using-declaration for non-member at class scope
SRC/route/connection_based_routing.h:115: error: expected ‘;’ before ‘=’ token
SRC/route/connection_based_routing.h:115: error: expected unqualified-id before ‘=’ token
In file included from SRC/draw/intra_logic_block.h:14,
                 from SRC/draw/draw.c:30:
SRC/route/route_tree_timing.h:40: error: ‘CBRR’ has not been declared
"

LemonPi, 2015-10-05:
Which directory are you using make from? I need more information to replicate it.

berbagci, 2015-10-05:
I'm using make in "vtr-verilog-to-routing-master" directory. After fixing some typos, now I'm getting the following error.

make[1]: `test_log' is up to date.
make[1]: `read_arch' is up to date.
make[1]: Nothing to be done for `all'.
fatal: Not a git repository (or any of the parent directories): .git
g++ -I../libarchfpga/include -Wall -Werror -Wpointer-arith -Wcast-qual -D__USE_FIXED_PROTOTYPES__ -ansi -pedantic -Wshadow -Wcast-align -D_POSIX_SOURCE -Wno-write-strings -D BUILD_VERSION=\"""\" -D BUILD_DATE=\""Mon, 05 Oct 2015 00:41:40 -0400"\" -D EZXML_NOMMAP -D_POSIX_C_SOURCE -std=c++0x -O3 -MD -MP -I-ISRC/util -ISRC/timing -ISRC/pack -ISRC/place -ISRC/base -ISRC/draw -ISRC/route -ISRC/power  -I../libsdc_parse/src -I../liblog/src -I/usr/include/freetype2   -ISRC/util -c SRC/route/route_common.c -o OBJ/route/route_common.o
SRC/route/route_common.c: In function ‘void mark_remaining_ends(int, const std::vector<int, std::allocator<int> >&)’:
SRC/route/route_common.c:647: error: expected initializer before ‘:’ token
SRC/route/route_common.c:649: error: expected primary-expression before ‘}’ token
SRC/route/route_common.c:649: error: expected ‘;’ before ‘}’ token
SRC/route/route_common.c:649: error: expected primary-expression before ‘}’ token
SRC/route/route_common.c:649: error: expected ‘)’ before ‘}’ token
SRC/route/route_common.c:649: error: expected primary-expression before ‘}’ token
SRC/route/route_common.c:649: error: expected ‘;’ before ‘}’ token
SRC/route/route_common.c: In function ‘void heap_::verify_extract_top()’:
SRC/route/route_common.c:1102: error: ‘constexpr’ was not declared in this scope
SRC/route/route_common.c:1102: error: expected ‘;’ before ‘float’
SRC/route/route_common.c:1104: error: no matching function for call to ‘std::vector<s_heap*, std::allocator<s_heap*> >::vector(<brace-enclosed initializer list>)’
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_vector.h:271: note: candidates are: std::vector<_Tp, _Alloc>::vector(std::initializer_list<_Tp>, const _Alloc&) [with _Tp = s_heap*, _Alloc = std::allocator<s_heap*>]
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_vector.h:257: note:                 std::vector<_Tp, _Alloc>::vector(std::vector<_Tp, _Alloc>&&) [with _Tp = s_heap*, _Alloc = std::allocator<s_heap*>]
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_vector.h:241: note:                 std::vector<_Tp, _Alloc>::vector(const std::vector<_Tp, _Alloc>&) [with _Tp = s_heap*, _Alloc = std::allocator<s_heap*>]
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_vector.h:227: note:                 std::vector<_Tp, _Alloc>::vector(size_t, const _Tp&, const _Alloc&) [with _Tp = s_heap*, _Alloc = std::allocator<s_heap*>]
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_vector.h:215: note:                 std::vector<_Tp, _Alloc>::vector(const _Alloc&) [with _Tp = s_heap*, _Alloc = std::allocator<s_heap*>]
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_vector.h:207: note:                 std::vector<_Tp, _Alloc>::vector() [with _Tp = s_heap*, _Alloc = std::allocator<s_heap*>]
SRC/route/route_common.c:1107: error: ‘begin’ was not declared in this scope
SRC/route/route_common.c:1107: error: ‘end’ was not declared in this scope
SRC/route/route_common.c:1108: error: expected primary-expression before ‘[’ token
SRC/route/route_common.c:1108: error: expected primary-expression before ‘]’ token
SRC/route/route_common.c:1108: error: expected primary-expression before ‘const’
SRC/route/route_common.c:1108: error: expected primary-expression before ‘const’
SRC/route/route_common.c:1116: error: ‘float_epsilon’ was not declared in this scope
make[1]: *** [OBJ/route/route_common.o] Error 1
make: *** [vpr] Error 2

kmurray, 2015-10-05:
Please try using a more recent version of GCC (e.g. 4.7 or later).

VPR assumes C++11 language features are supported by the compiler.  It looks like [GCC 4.4](https://gcc.gnu.org/projects/cxx0x.html) doesn't support some of the features VPR uses.

Our [buildbot](http://islanders.eecg.utoronto.ca:8080/waterfall) currently builds against GCC 4.7.  So GCC 4.7 or later should build correctly.

closed 2016-01-22 by kmurray