verilog-to-routing/vtr-verilog-to-routing #111: VPR can't handle signals that enters both clock ports and normal input ports

EliasVansteenkiste, 2016-01-19:
VPR fails while trying to read the blif file which describes the LU_Network design from the Titan benchmark suite.

VPR exits with 75 identical error messages.
I included all the error messages in the attachmment. Here is the first one as an example:
```
Error 1: You have a signal that enters both clock ports and normal input ports.
Input net for block ddr:ddr_inst|ddr_0002:ddr_inst|ddr_p0:p0|ddr_p0_memphy:umemphy|ddr_p0_read_datapath:uread_datapath|ddr_p0_flop_mem:read_buffering[5].read_subgroup[0].uread_fifo|data_stored[0][0] #38441 is net (null) #-1 but connecting net is ddr:ddr_inst|ddr_0002:ddr_inst|ddr_p0:p0|ddr_p0_memphy:umemphy|ddr_p0_new_io_pads:uio_pads|ddr_p0_altdqdqs:dq_ddio[5].ubidir_dq_dqs|altdq_dqs2_ddio_3reg_stratixiv:altdq_dqs2_inst|dqsbusout #212424.
```
[LU_network_error.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/95120/LU_network_error.txt)

EliasVansteenkiste, 2016-01-22:
After checking out the latest version of VTR (Revision: d71ee16-dirty)
I am able to pack directrf, but for LU_Network the error messages persist.

I attached the log file for LU_Network.
[vpr_stdout_LU_Network.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/100813/vpr_stdout_LU_Network.txt)

kmurray, 2017-03-11:
Should be fixed in 6ccf516.

We handle these cases when cleaning up the BLIF netlist after parsing, but before packing. The packer algorithm inherently does not support mixed clock/data nets; likely due to pin counting issues (?).

We split a clock's data sink pins into a new net driven by a `.latch` (with no data input) clocked by the original net.  This is not a functionality preserving transformation, but it is not clear how this should be handled otherwise.

closed 2017-03-11 by kmurray