verilog-to-routing/vtr-verilog-to-routing #114: VPR Track Direction Visualization Error

rbkettlewell, 2016-01-21:
There seems to be a bug with the VPR graphics enabled whereby, in some cases, unidirectional routing tracks have the direction arrows reversed to what would be expected. I am seeing this behaviour with Wilton switches, but have not retested with subset or universal options. Also, I am using a daily build from last the end of last week (I'm not aware of a versioning flag location in the source directory to confirm the exact build number)

I've attached a picture that shows the routing originating from a outpad pin connecting to a CLB input, however the switch direction arrow appears reversed in an intermediate segment. I've also attached the architecture file with the image. 

[image]
[troll_arch8.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/99555/troll_arch8.xml.txt)

kmurray, 2016-01-21:
Thanks for the report.

It does appear that the arrows are drawn in the wrong direction inside the switch block.

Could you provide the procedure (e.g. blif netlist and command line) to reproduce the issue?

rbkettlewell, 2016-01-21:
Sure, though I don't believe I've seen a visualization yet (using multiple different .blif files) where the switch matrices don't have at least a few incorrect arrows.

Here is my command line entry in a subfolder of vpr : 
../vpr troll_arch.xml Combinational.blif --route_chan_width 12

*** Note: I have seen the same problem without fixing the routing width

[Combinational.blif.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/99927/Combinational.blif.txt)

Thanks for taking a look.

kmurray, 2017-03-10:
Fixed in e66c0ac. Drawing edges from X to Y channels was not being treated the same as Y to X channels which caused the direction to be inverted. This is now handled correctly.

closed 2017-03-10 by kmurray