verilog-to-routing/vtr-verilog-to-routing #115: VPR fails placing gaussianblur benchmark design.

EliasVansteenkiste, 2016-01-23:
Right before placing gaussianblur, VPR crashes with the following assertion:
vpr: SRC/base/read_netlist.c:1148: void load_atom_index_for_pb_pin(t_pb_route*, int): Assertion `driver != OPEN' failed.

kmurray, 2016-01-25:
Thanks for the report Elias.  I'm working to reproduce the problem.

kmurray, 2016-01-26:
Which revision of VPR are you using?

I've been able to pack and start placement (with no crash) using f2a0261.

EliasVansteenkiste, 2016-01-27:
I used revision d71ee16.
VPR fails after autosizing the circuit. Nothing is written to stderr.
I attached the log file.
I will try again after I pull in the newer version of vtr.
[gaussianblur.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/107082/gaussianblur.txt)

EliasVansteenkiste, 2016-01-28:
The same thing happens for revision 960e4e7.

kmurray, 2016-01-28:
Odd. Have you been re-running both packing and placement, or placement only?

One of the fixes I put in recently fixed a corruption issue when writing the post-packing netlist (.net) file. 

If you've been doing placement only, try re-packing with one of the new revisions first.

kmurray, 2017-08-01:
Confirmed working with latest trunk.

closed 2017-08-01 by kmurray