verilog-to-routing/vtr-verilog-to-routing #119: installing VTR

fatma9, 2016-03-05:
hello, i'm facing a problem in installing VTR and this is the error:

OBJ/simulate_blif.o: In function `initialize_pin':
/vtr/ODIN_II/SRC/simulate_blif.c:1377: undefined reference to `set_pin_cycle'
OBJ/simulate_blif.o: In function `create_lines':
/vtr/ODIN_II/SRC/simulate_blif.c:2398: undefined reference to `is_clock_node'
OBJ/simulate_blif.o: In function `write_wave_to_modelsim_file':
/vtr/ODIN_II/SRC/simulate_blif.c:2868: undefined reference to `is_clock_node'
OBJ/simulate_blif.o: In function `print_update_trace':
/vtr/ODIN_II/SRC/simulate_blif.c:3600: undefined reference to `get_pin_cycle'
collect2: error: ld returned 1 exit status
Makefile:84: recipe for target 'odin_II.exe' failed
make[1]: *** [odin_II.exe] Error 1
Makefile:12: recipe for target 'ODIN_II' failed
make: *** [ODIN_II] Error 2

any help please

kmurray, 2016-03-05:
What version of VTR are you using (release, trunk etc.)?

What is your operating system, and compiler version?

fatma9, 2016-03-05:
thanks for answering  
VTR : vtr release 7.0 
operating system: ubuntu version 4.2.0-30-generic
compiler : gcc (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010  Copyright (C) 2015 Free Software Foundation, Inc.

kmurray, 2016-03-06:
I've been able to reproduce the issue with the VTR 7.0 release on Ubuntu 15.04, with gcc 5.1.1
New versions of GCC (i.e. >= 5.0) do not like how ODIN II is defining some inline functions, generating a bunch of warnings about inline functions being declared but not defined.

I was able to build the VTR 7.0 release does build with GCC 4.9, can you try building with a version of GCC < 5.0 ?

fatma9, 2016-03-07:
I installed it successfully , I used ubuntu 14.04.4 and gcc 4.8.4
thank you for your help.

closed 2016-03-07 by fatma9

nitzzz6, 2017-08-01:
Hi, I am having the same issues. I tried using the above same versions of Ubuntu and gcc. Please let me know what corrections should I make.

kmurray, 2017-08-01:
What is the output when you run `gcc --version` from the command line?

nitzzz6, 2017-08-01:
I have tried using gcc versions 4.6 and 5.4. Neither of them are working.
What version do I use and also what Ubuntu version to use?

Thanks in advance.

kmurray, 2017-08-01:
As noted above:
> ubuntu 14.04.4 and gcc 4.8.4

are known to work with the VTR 7 release.

jeanlego, 2017-08-01:
Did you install all the required package? 
Run_on_arm text file list all the dependency and fixes to run vtr.( It says arm but it is the same as x86 architecture.

Try to follow that .

Also clang works with vtr if you want to give that a try

fatma9, 2017-08-02:
hi ! i changed the version of ubuntu and gcc and it worked successfully. i used ubuntu 14.04.4 and gcc 4.8.4