verilog-to-routing/vtr-verilog-to-routing #120: Place&route simple overlay (VPR) -- any doc?

robertorigamonti, 2016-03-07:
Hi,

I would like to use VPR in a project I'm working on, since it looks brilliant to me.
I've spent the past three days trying to get a simple example working, but I've managed to do very little. Below are the files I've been able to produce.
My final goal is to have an overlay architecture like the one implemented in "A high-performance overlay architecture for pipelined execution of data flow graphs", but simplified so that each node is a simple adder -- or anything else, I'm interested in the place&route part only!

In my tests, what I've finally achieved to do is to have four CLB blocks of which just two (I would have loved to have all four, but couldn't get it work) are adders -- it might look miserable to you, but I've spent the rest of the time getting errors, so it's kind of a huge step for me :S.
The system reacts in a very strange way to changes in the model/architecture --- for instance,
 sometimes the four blocks are represented as a single one, sometimes it simply gives errors...

I've searched for hours on the net to see if any documentation was present --- besides the architecture reference (which has some errors) and the tutorial, both of which are incomplete and present something which is way more difficult than what I'm aiming too.

Is there any other document/forum/tutorial I can refer to?
In particular, how can I specify that I want, for instance, a 3x3 grid of blocks in the architecture, and then select the blocks (0,0), (1,0), and (2,2) and perform operations with them, linking for example the output of (0,0) to one of the inputs of (1,0)?
Thanks a lot in advance!
Rob

[arch2.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/161123/arch2.xml.txt)
[model.blif.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/161124/model.blif.txt)

robertorigamonti, 2016-03-07:
Ok, I've managed to select the size (oups, missed that option in device... :S) but still I cannot impose that each io is either an input, an output, or both input and output but not two inputs...

kmurray, 2016-03-07:
I assume you've seen the VTR documentation at: https://vtr.readthedocs.org/

### Device Size
To set the device grid size you can use the [layout tag](https://vtr.readthedocs.org/en/latest/arch/reference/#global-fpga-information).

### Bi-directional IOs
Your architecture already models bi-directional IOs:
```
    <pb_type name="io" capacity="8">
      <input name="outpad" num_pins="1"/>
      <output name="inpad" num_pins="1"/>

      <mode name="inpad">
        <pb_type name="inpad" blif_model=".input" num_pb="1">
          <output name="inpad" num_pins="1"/>
        </pb_type>
        <interconnect>
          <direct name="inpad" input="inpad.inpad" output="io.inpad">
            <delay_constant max="4.243e-11" in_port="inpad.inpad" out_port="io.inpad"/>
          </direct>
        </interconnect>
      </mode>

      <mode name="outpad">
        <pb_type name="outpad" blif_model=".output" num_pb="1">
          <input name="outpad" num_pins="1"/>
        </pb_type>
        <interconnect>
          <direct name="outpad" input="io.outpad" output="outpad.outpad">
            <delay_constant max="1.394e-11" in_port="io.outpad" out_port="outpad.outpad"/>
          </direct>
        </interconnect>
      </mode>
      ...
      <gridlocations>
        <loc type="perimeter" priority="10"/>
      </gridlocations>
  </pb_type>
```
Here you define a top level ``pb_type`` named ``io``, which will be located around the perimeter of the device. 

The architecture description uses ``mode``s to describe specific operational configurations for a block.  

Your ``io`` block supports two ``mode``s: ``inpad`` and ``outpad``.  VPR's packer will choose the appropriate mode for each ``io`` block it uses.

Also take note of the ``capacity="8"`` attribute on the ``io`` pb_type, this indicates that there will be 8 ``io`` blocks at each grid location.

### Logic Block

Another thing to be careful with is the `num_pb` attributes.

`num_pb` specifies how many instances of the current `pb_type` exist in the parent `pb_type`. So in your architecture file:
```
    <pb_type name="clb">
      <input name="I1" num_pins="3" equivalent="true"/>
      <input name="I2" num_pins="3" equivalent="true"/>
      <output name="O" num_pins="3" equivalent="false"/>

      <pb_type name="ble" blif_model=".subckt sum" num_pb="3">
        <input name="a" num_pins="1"/>
        <input name="b" num_pins="1"/>
        <output name="out0" num_pins="1"/>
      </pb_type>
      ...
    <pb_type/>
```
``num_pb="3"`` Indicates that each top-level `clb` block consists of 3 `sum` subckts.  From your description above it seems like you want a single `sum` per `clb` block (i.e. `num_pb="1"`).

You also need to take care to define the interconnect between parent and child pb_types (see [here](https://vtr.readthedocs.org/en/latest/arch/reference/#interconnect) for what the different tags mean).  If the interconnect is not correctly specified then VPR will be unable to pack your circuit.

### Documentation

We have limited resource but are always looking to improve our documentation, so if you have specific suggestions/enhancements please let us know.

robertorigamonti, 2016-03-08:
Thanks a lot for your detailed answer!! :)
What I've posted was probably not the best set of files to explain my problems (they were just the files I've been able to obtain after playing a bit to explore the different options, and not something aimed at getting the overlay I'm targeting -- I should have tailored them before posting!).
Yesterday I've played some more, and I've managed to get something more.
However, I still cannot understand how to achieve the following points:
- concerning the I/O pins, my problem is that I'd like to have at most one input pin and at most one output pin for each side of the CLB, but if I set the capacity to "1" then I'll get either an input or an output, while if I set it to 2 I sometimes get two inputs (or outputs). I've removed the modes and tried to add just an inpad and an outpad, setting then the capacity to "1", but both visually and in the .place file only the input or the output pin is present (the other one "disappears")
- I am not able to impose the topology of the connections --- I'd like to have a Manhattan-style grid with all the inputs and the blocks connected by horizontal and vertical lines only, and with connections that can "traverse" blocks by setting muxes in the blocks on the path (what I mean is that, if I have three blocks arranged in a row as 1-2-3, to connect 1 with 3 I have to set the mux in 2 that links its west input with its east output). In my attempts, I have configured all the interconnections inside a CLB using the mux keyword and "muxing" all the possible inputs on each output, then I guessed I had to use a switch_block of type "custom" and consequently set some switchfuncs in a switchblocklist (mostly by randomly copying some examples on the net without understanding much...), but I still get diagonal connections, as can be seen in the figure (here after a "place" step).
[image]
- Again for the point above, I cannot impose direct connections. If my CLB is as follows:
```
    <pb_type name="clb">
      <input name="I" num_pins="4"/>
      <output name="O" num_pins="4"/>

      <pb_type name="FU" blif_model=".subckt FU" num_pb="1">
        <input name="a" num_pins="1"/>
	<input name="b" num_pins="1"/>
	<output name="out" num_pins="1"/>
      </pb_type>

      <interconnect>
	<mux name="muxN" input="clb.I[1] clb.I[2] clb.I[3] FU.out" output="clb.O[0]"/>
	<mux name="muxS" input="clb.I[0] clb.I[2] clb.I[3] FU.out" output="clb.O[1]"/>
	<mux name="muxW" input="clb.I[1] clb.I[0] clb.I[3] FU.out" output="clb.O[2]"/>
	<mux name="muxE" input="clb.I[0] clb.I[1] clb.I[2] FU.out" output="clb.O[3]"/>
	<mux name="muxFU1" input="clb.I[0] clb.I[1] clb.I[2] clb.I[3]" output="FU.a"/>
	<mux name="muxFU2" input="clb.I[0] clb.I[1] clb.I[2] clb.I[3]" output="FU.b"/>
      </interconnect>

      <fc default_in_type="frac" default_in_val="0.15" default_out_type="frac" default_out_val="0.10"/>

      <pinlocations pattern="spread"/>
      <gridlocations>
        <loc type="fill" priority="1"/>
      </gridlocations>
    </pb_type>
```
and I want, for instance, to link the south output of the CLB above with the north input of the CLB below, I would expect to do so by using:
```
  <directlist>
    <direct name="directNS1" from_pin="clb.O[0]" to_pin="clb.I[1]" x_offset="0" y_offset="-1" z_offset="0"/>
  </directlist>
```
However, vpr complains about this:
`Error 1: [LINE 145] Invalid pin - clb.O[0], name should be in the format "pb_type_name"."port_name" or "pb_type_name"."port_name[end_pin_index:start_pin_index]". The end_pin_index and start_pin_index can be the same.`

Could you please help me again with these points?

Concerning the documentation, I unfortunately know too well the "lack of resources" problem :(
IMHO it is excellent in many aspects, what's probably missing is a full, **simple** example (what I did so far could be one of those examples - I am more than willing to give you the code to enrich the documentation (once it works...)) and possibly a "dumb-proof" explanation of the arcane parts --- I have a CS and not an EE background, though I have studied quite a lot of electronics and enjoy it a lot. I know that for someone in the field reading "wireconn from_type="l4"" is perfectly clear, but for me this "l4" falls down the blue sky, and googling for it is not that trivial. The link with BLIF is another weird thing --- to make it work I've tested all the possible "combinations" and tried to reverse engineer the code... Even more confusing is the "input called outpad - output called inpad" stuff --- there is for sure a trivial explanation for that, but without knowing it its a good source of headaches!

Thanks again and have a nice day!
Rob

opetelin, 2016-03-08:
Hi Rob, 

I can't comment on the directs issue, but regarding the other two, what VPR displays changes based on what phase of the tool flow you are in. So during the placement step what you see is a kind of "netlist" view that shows you which FPGA blocks should be connected, but not necessarily how they should connect -- rest assured though that the routing architecture you specified is still there (and will be used during the subsequent routing step). So:

1. Each I/O block in your architecture has one input and one output pin. But the placer marks each block for only input or output use -- the other pin is still there though.

2. The Manhattan (island-style) grid of interconnect is there. It is used by the router to implement the higher-level netlist connections determined by the placer (shown in the picture you attached). I would recommend against using the "custom" switch_block type as this is an advanced feature that allows you to control the precise switch pattern inside the switch block structures; the "wilton" switch block type should work fine.

3. I think the directs issue might just come down to formatting. Try "clb.I[0:0]" and "clb.O[0:0]" instead.

Once you get through the entire place and route flow, you will be able to see the detailed routing architecture using VPR graphics (press the "Toggle RR" button a few times). Also, if you're interested in FPGA routing architecture, check out "Flexibility of Interconnection Structures for Field-Programmable Gate Arrays" (Rose, Brown, 1991) and "Architecture and CAD for Deep-Submicron FPGAs" (Betz, Rose, Marquardt, 1999) for a more detailed reference.

robertorigamonti, 2016-03-08:
Thanks for your answer, Opetelin. I will read the references you pointed me.

Just to explain a bit more my objectives, what I would like to obtain is similar to what presented in this picture:
[image]
There, I have on each side 1 input and 1 output pin. However, if I put in my code the following:
```
    <pb_type name="io" capacity="1">
      <input name="outpad" num_pins="1"/>
      <output name="inpad" num_pins="1"/>

      <pb_type name="inpad" blif_model=".input" num_pb="1">
        <output name="inpad" num_pins="1"/>
      </pb_type>
      <pb_type name="outpad" blif_model=".output" num_pb="1">
        <input name="outpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct name="outpad" input="io.outpad" output="outpad.outpad"/>
    	<direct name="inpad" input="inpad.inpad" output="io.inpad"/>
      </interconnect>

      <fc default_in_type="frac" default_in_val="0.15" default_out_type="frac" default_out_val="0.10"/>

      <pinlocations pattern="custom">
        <loc side="left">io.outpad io.inpad </loc>
        <loc side="top">io.outpad io.inpad </loc>
        <loc side="right">io.outpad io.inpad </loc>
        <loc side="bottom">io.outpad io.inpad </loc>
      </pinlocations>

      <gridlocations>
        <loc type="perimeter" priority="10"/>
      </gridlocations>
    </pb_type>
```
(which expresses what I have in mind) and I use the following example BLIF file:
```
.model top
.inputs ta tb tc td
.outputs to tp tq tr
.subckt FU a=ta b=tb out=to
.subckt FU a=to b=tc out=tp
.subckt FU a=ta b=tc out=tq
.subckt FU a=ta b=tc out=tr
.end

.model FU
.inputs a b
.outputs out
.blackbox
.end
```
my inputs at the end of the place and route (that is, after all the VPR steps) are simply cut away:
```
Netlist file: mod3.net   Architecture file: arch3.xml
Array size: 3 x 3 logic blocks

#block name	x	y	subblk	block number
#----------	--	--	------	------------
to		1	1	0	#0
tp		1	2	0	#1
tq		2	1	0	#2
tr		2	2	0	#3
out:to		1	0	0	#4
out:tp		0	2	1	#5
out:tq		2	0	1	#6
out:tr		4	2	0	#7
```
The same holds if a put capacity="2". What I am looking for is to have both the "mode" as traditionally done, but once one of the two is used, it cannot be reused even if the capacity is 2.

For the connections, I would prefer to avoid the "custom" option, but the switch blocks will otherwise add the connections that link blocks in diagonal, whereas I need them to be connected with their 4-neighborhood as in the picture. Diagonal connections have to be achieved by passing through free output ports of neighbours. Is there an easy way to impose this constraint?

Concerning point (3) you're right, I had tried it too but it didn't work, now it works if I put the [0:0], possibly it was another mistake in the code that prevented it from working... However, I am still not able to see this connections when clicking on "toggle RR" (I just see the "standard" ones --- maybe I have to play a bit more to get something done here...)

opetelin, 2016-03-08:
I see, so if I understand correctly, what you're trying to do is implement a systolic array of sorts where each compute unit connects to its nearest four neighbours. In the paper you referenced, the authors overlaid these synthesized compute units on top of the existing FPGA fabric (Stratix IV in that case). Just to make sure, are you looking to create an overlay on top of the existing FPGA fabric like in the paper, or do you want to actually change the underlying FPGA hardware?

Cheers,
Oleg

robertorigamonti, 2016-03-08:
Well, I have written a simple Qt simulator that implements a simplified version of the approach they use in the paper (but with the same structure), and now I would like to "configure" it on the fly starting from a model --- that is, I have for instance the graph described in the mod3.blif file shown above and I want to place each block in my simulated environment and fire on the appropriate connections. From this perspective, I don't really care if there's an FPGA below or what, I just have a matrix of blocks interconnected as in the paper, and I need to know that for a given problem I have to put (for example) block 1 in cell(1,2), block 2 in cell(1,3) and then activate the proper connections.
When I saw VPR I realized that it implements exactly what I am looking for, but I have issues "forcing" it to work on my idealized structure...

opetelin, 2016-03-08:
Hi Rob, 

With regard to changing the FPGA hardware, VPR performs intra-logic-block and inter-logic-block routing separately. The connections inside the logic block are routed sometime during the packing phase, while the connections between logic blocks are implemented during the routing phase after placement. The rr_graph (routing resource graph) that VPR generates for the router does not include nodes/edges that specify the internal connectivity of the CLBs. To implement what you want, you would probably have to modify the rr_graph generation code (build_rr_graph in vpr/SRC/route/rr_graph.c) to insert graph edges from input to output pins in accordance with your architecture. The VPR placer should have some awareness of your direct-only connectivity, and it will try to optimize for it, but the placer was not written for this kind of architecture and you might have to play with that code too.

I can't comment on what floor-planning functionality VPR has right now, but if you'd rather go with the overlay route like in the reference paper, it may be easier to follow their tool flow as well.

robertorigamonti, 2016-03-08:
Ok, got it, I'll try to have a look at the code and see what I can do.
Thanks a lot for your help!!! :)

closed 2016-07-20 by kmurray