verilog-to-routing/vtr-verilog-to-routing #121: Cannot get past VPR stage

GeorgeAthanasopoulos, 2016-03-09:
This is not a bug I am sure however I don't know how else to find answers. I tried reading the manual that you have posted but I did not find or understand why I cannot get past VPR.

My OS is ubuntu 14.04
My VTR version is the current latest

The run_vtr_task.pl basic_flow worked as it was intended..

I tried to execute a few vtr flows

The command took me a while to figure out how to use but in the end I got it working . First I picked a circuit from verilog folder in benchmarks called bgm.v then an architecture randomly from both timing and no_timing . To give an example of the command I used so u can get an idea

./scripts/run_vtr_flow.pl benchmarks/verilog/raygentop.v arch/no_timing/fracturable_lut_sweep/k6_N8_I27_fleI6_fleO1_ff1_nmodes_1.xml

This was one command I used in order to execute it while i was at vtr_flow folder

the result i got was : 

k6_N8_I27_fleI6_fleO1_ff1_nmodes_1/raygentop...failed: vpr

I tried with different architectures and verilog files but the result was always the same

I removed VTR and reinstalled it again in case something was wrong there but the problems are reproduced.

I tried the same with blif files as well but later realised they are netlists so it was only normal that ODIN would fail right? So i tried skipping ABC and ODIN II parts while  using -starting_stage VPR on which the result was the same.

Apparently I understood something wrong and I probably lack the background in hardware to understand if this has something to do with incompatible architectures used on circuits but after trying a few i didn't know what to do.

My goal was to use VTR for my thesis and try and implement a parallel processing version of the placement algorithm and compare my results to the original serial execution results both in quality of circuit and speedup. However in order to do that I have to run benchmarks on both and I fail at it at the moment . Any help would be greatly appreciated. If i contacted the wrong area it's because I have no idea how else to contact the people responsible for this project and if there's somewhere else I d rather contact you please kindly let me know.

Thanks in advance

George

opetelin, 2016-03-09:
This was a bug. When you use the "run_vtr_flow" script, a "temp" directory is created with tool outputs. Looking at temp/vpr.out I got:

```
...../vtr/vtr_flow/../vpr/vpr k6_frac_N10_mem32K_40nm.xml raygentop --blif_file raygentop.pre-vpr.blif --timing_analysis on --timing_driven_clustering on --cluster_seed_type blend --routing_failure_predictor safe --sdc_file --seed 1 --nodisp
Error 1: 
Type: Other
File: SRC/base/ReadOptions.c
Line: 811
Message: Unexpected token '1' on command line.
```

Looks like the `--sdc_file` option was being given to VPR without an actual SDC file. 

I've pushed a fix to the master branch.

GeorgeAthanasopoulos, 2016-03-09:
Thank you for the quick reply.

opetelin, 2016-04-08:
Hi George, you mentioned in the other thread that run_vtr_flow.pl is still giving you trouble.

Did the pushed fix above work for you? If not, can you post the error that you're getting during the VPR stage as well as the command line arguments that you invoke the script with? (you can find the "vpr.out" file in a "temp" directory created when you run the script)

jgoeders, 2016-04-08:
The issue here is that you are using a "no_timing" architecture file, without disabling timing analysis.  If you pass the '-no_timing' option to run_vtr_flow it should work.

GeorgeAthanasopoulos, 2016-04-09:
Thank you. It seems it is getting past the stage that it used to fail at now. I would like to ask a couple of things more though.

I have trouble enabling graphics as I don't quite understand what do I need to change. 
I tried going to the makefile of VPR and change the following 2 lines

I commented out this : 
#FLAGS = $(OPT_FLAGS) $(INC_FLAGS) -D EZXML_NOMMAP -DNO_GRAPHICS -D_POSIX_C_SOURCE 
and uncommented this:
LIB = -lm -lX11 -lvpr_6

as suggested at the top of the makefile.

The result was when i used make on vtr_release folder, VPR was not compiled so I can only presume that I am doing something wrong

I tried running the following on the non-graphic enabled version:
./vtr_flow/scripts/run_vtr_flow.pl vtr_flow/benchmarks/verilog/bgm.v vtr_flow/arch/timing/k6_N10_memDepth16384_memData64_40nm_timing.xml

It seemed to be working but after 40 minutes i stopped it. Since I have never succesfully run something like this I would like to ask what is the expected time of execution ? I understand this is hardware related but for an I5 processor with 16 giga ram should that take something like minutes? Hours? Days? any estimation would help a lot so I know if i should let it continue on or not.

Finally I would like to ask if there's anywhere  an API for this programm that I can find in order to understand the structs used to represent data and alter them wherever needed.

Thank you 

George

P.S. Sorry again if I am asking things that might seem obvious to you. If I should check something else or contact other people about them I would really appreciate it if you could point me to the right direction.

GeorgeAthanasopoulos, 2016-04-09:
Ignore the how long should it be running. I verified it is running as i managed to succesfully finish the placement and routing of the example I mentioned in my second post with the -no_timing parameter.

It seems that timing takes a huge amount of time in comparison ? or maybe the architectures differ too much

./vtr_flow/scripts/run_vtr_flow.pl vtr_flow/benchmarks/verilod/bgm.v vtr_flow/arch/no_timing/memory_sweep/k4_N10_memSize512_memData2.xml -no_timing

took 21 minutes 38 seconds to execute VPR or 1298 seconds if you prefer

kmurray, 2016-04-11:
**Documentation**
For documentation on VTR and VPR see: https://vtr.readthedocs.org/.

**Enabling Graphics**
Set `ENABLE_GRAPHICS = true` in vpr/Makefile.  This requires some X11 related libraries to be installed.  Also see [here](https://vtr.readthedocs.org/en/latest/vpr/compiling/).

**Quality of Results**
VPR produces most of its QoR metrics in the log file.  You can parse these manually or use the [parse_vtr_flow](https://vtr.readthedocs.org/en/latest/vtr/parse_vtr_flow/) script.

**API**
There is no formally defined API for vpr.  However the main data structures are described with reasonably detailed comments.

jgoeders, 2016-04-11:
Kevin, perhaps we should add the -no_timing flag to the documentation here:  https://vtr.readthedocs.org/en/latest/vtr/run_vtr_flow/

Also, right now if VPR is run with timing analysis on a non-timing benchmark, it segfaults.  Sorry, I can't remember exactly where that happens, but it is actually within some error handling code.  I was helping another student with the same issue the other day, but I don't have it set up on my system currently to track down the problem.

kmurray, 2016-04-11:
I've updated the documentation to include the -no_timing option.

I've also updated the error check so it no longer segfaults -- although the program still exits due to an invalid timing graph. 

I'm planning to look at the timing analysis code over the summer, so hopefully issues like this will be cleaned-up then.

closed 2016-05-10 by jgoeders