verilog-to-routing/vtr-verilog-to-routing #123: problem when using --fix_pins option

berbagci, 2016-03-20:
I'm running VPR with random pad locations and it works fine but when I use the option, --fix_pins "file.pads", VPR crashes with the following error messages:
also attached the VPR logfile 

...
[logfile.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/181173/logfile.txt)

vpr: SRC/route/route_timing.c:530: bool timing_driven_route_net(int, int, float, float, float, float, float, CBRR&, float*, int, t_rt_node**, float*, t_slack*): Assertion `rr_node[rt_root->inode].get_occ() <= rr_node[rt_root->inode].get_capacity()' failed.
Abort (core dumped)

kmurray, 2016-03-21:
This assertion failure appears to indicate that the source (i.e. driver) of a net is becoming congested during routing (which the code is assumes to be impossible).

Since you only see this problem when fixing I/O pad locations my best guess is that there is a problem with the I/O constraints. Perhaps two or more pins are assigned to the same location?

berbagci, 2016-03-21:
Hmm I see. I did double-check but couldn't see any issues with the IO constraint file. The thing is I'm trying to force all the inputs on one side (i.e., left). You think this might be the issue?

berbagci, 2016-03-21:
Actually, I tried another benchmark that has only 4 input pins and one output pin and vpr failed again (with the same error message). I think there might be a problem with --fix_pins option.

kmurray, 2016-03-21:
The --fix_pins is not used very often in the standard flow, so it certainly possible there is a latent bug.

There was a problem with this option which was fixed in #117.  Could you try with the latest trunk version of VPR?

berbagci, 2016-03-21:
I tried the latest trunk version but the problem is still there.
I'm building an FPGA fabric and post-processing VTR placement, route and
net files to generate the bitstream for my FPGA.
I just wanted to make sure whether forcing I/O pads to specific locations
causing the problem (not enough wire resources etc.) or --fix_pins option
is not working. I'm not sure maybe it's one of them or both. But I suspect
the option itself might be buggy. I tried looking into the code but it's
complicated. Is there anyone who is currently working on the code so that I
can talk to him about the possible fix?
Thanks

Burak

kmurray, 2016-03-21:
If you can provide the architecture XML and blif file I can try to reproduce the issue and try to track down the cause.

However I can't make any timeliness commitments for a fix.

berbagci, 2016-03-21:
Ok, sure. I'm sending you the files.

Thank you so much for your time,

Burak

berbagci, 2016-03-22:
As I took a look at the code (place.c), I realized that for --fix_pins random option, the placement for IOs is done at the very beginning and they are marked as fixed and this works fine. However, when a user defined pad file is being used, this initial placement of IOs is skipped and is done later. Maybe the problem stems from this fact. Maybe IOs should be placed initially (as in the random case)?

kmurray, 2016-03-23:
From my reading of the code the user specified placement is applied last (and the blocks are marked as fixed).  

Can you provide a test case to reproduce the issue? Without it I can only speculate.

eshellko, 2016-06-24:
I have same problem with fixed pin locations.
After some researches, it appears that 3-rd index of location not assigned to placed block.
It is fixed in file 'read_place.c'. Function 'read_user_pad_loc'. Added assignment to block[bnum].z.
```
		block[bnum].x = i; /* Will be reloaded by initial_placement anyway. */
		block[bnum].y = j; /* I need to set .x only as a done flag.         */
block[bnum].z = k; /* added 3rd index here */
		block[bnum].is_fixed = true;
```
This is just my assumption. I'am not completely sure it will work for everyone.
I have checked it only for 2 different designs.
So, will wait for approval.

kmurray, 2016-07-20:
Thanks to eshellko's patch this should now be resolved.

closed 2016-07-20 by kmurray