verilog-to-routing/vtr-verilog-to-routing #125: Error in running run_vtr_flow.pl

aliyah3, 2016-04-08:
I tried the command below and get this error:

[...@nano scripts]$ perl run_vtr_flow.pl bgm.v k6_N10_I33_Fi6_L1_frac0_ff1_45nm.xml
Circuit file not found (bgm.v) at run_vtr_flow.pl line 185.

is this the right way of running it?

GeorgeAthanasopoulos, 2016-04-08:
Hm i think u have used the paths of the files wrong

For example for me 

./vtr_flow/scripts/run_vtr_flow.pl vtr_flow/benchmarks/verilog/bgm.v vtr_flow/arch/no_timing/memory_sweep/k4_N10_memSize512_memData2.xml

when i execute this from the vtr_release folder

Unfortunately i cannot seem to make it get past VPR stage and it always fails regardless of the benchmark i use or the architecture.

jgoeders, 2016-04-08:
George, the issue here is that you are using a "no_timing" architecture file, without disabling timing analysis.  If you pass the '-no_timing' option to run_vtr_flow it should work.

closed 2016-05-10 by jgoeders