verilog-to-routing/vtr-verilog-to-routing #128: running vtr task with additional script option for standard script

werneazc, 2016-05-02:
Hello,

during me benchmarking I had a problem using a additional parameter. I set up a script like this:

> circuits_dir=/home/user/coding/EXTRA/extra-granularity-bench/verilog

> archs_dir=/home/user/coding/EXTRA/extra-granularity-bench/arch

> circuit_list_add=stereovision3.v

> arch_list_add=K05_N10_40nm_subset.xml

> arch_list_add=K05_N10_40nm_wilton.xml

> parse_file=extra_standard.txt

> script_params=-max_router_iterations 25

Using this definition, run_vtr_flow starts VPR with the parameter "-temp_dir . ", but that parameter is not know by VPR and the benchmark fails. If one don't use any parameter, then the script executes well. I'm not familiar with Perl so I wasn't able to figure out why his is happen or to fix that bug.

Best regards

kmurray, 2016-05-02:
This was indeed a bug in run_vtr_trask.  It should now be fixed in 475308c.

The only other change I had to make was to provide:
> script_params=-vpr_max_router_iterations 25
(note the vpr_ prefix).

This prevents the --max_router_iterations option being specified multiple times when VPR is called.

closed 2016-05-02 by kmurray

werneazc, 2016-05-02:
I have one last question. If I use

> -vpr_max_router_iterations 25

I have a difference in two vpr output files.

**vpr.out** shows the following line:

> home/andrewerner/tools/vtr-verilog-to-routing/vtr_flow/../vpr/vpr K05_N10_40nm_subset.xml stereovision3 --blif_file stereovision3.pre-vpr.blif --timing_analysis on --timing_driven_clustering on --cluster_seed_type blend --routing_failure_predictor safe --seed 1 --nodisp

while **vpr_stdout.log** shows that line:

> /home/andrewerner/tools/vtr-verilog-to-routing/vtr_flow/../vpr/vpr K05_N10_40nm_subset.xml stereovision3 --route --blif_file stereovision3.pre-vpr.blif --route_chan_width 28 --max_router_iterations 25 --cluster_seed_type blend --nodisp --gen_postsynthesis_netlist off

So whats the differenz between that two outputs. Why they exits and whats the one I have to use to collect my benchmark results for instance area estimation.

Regards
André

jgoeders, 2016-05-02:
The run vtr_flow script executes vpr twice. 

The first execution performs packing, placement, and routing.  The routing phase actually consists of many routing attempts in order to determine the minimum channel width for which routing will succeed.   The output of this will be in **vpr.out**.

When vpr is executed the second time, it is done to only performing routing and timing analysis with a relaxed channel width.  The minimum channel width from the first execution is increased by 30%, and routing is performed again (That is why you see the argument --route_chan_width 28).  The output of this will be in **vpr.crit_path.out**.  

I believe **vpr_stdout.log** just shows the stdout of the most recent execution, so it should be the same as **vpr.crit_path.out** in this case.

werneazc, 2016-05-03:
Thank's again for that fast and considerable reply.

Regards
André