verilog-to-routing/vtr-verilog-to-routing #129: Error with power estimation in VPR

Ahmad90, 2016-05-08:
Hi,

I create all requirements for VPR 7.0 power estimation, and i could place and rout my architecture, but for power estimation i get this Error:

ERROR(1): [LINE 9] Node 'power' has invalid property method="auto".

Why VPR can not recognize all of power methods?

[image]

tnx...

jgoeders, 2016-05-08:
Could you attach the architecture file, so I can take a look at it?

Ahmad90, 2016-05-08:
yes, tnx a lot...

Ahmad90, 2016-05-08:
yes, tnx a lot...

[k1_N4_90nm.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/253983/k1_N4_90nm.txt)

jgoeders, 2016-05-10:
The **method** property only applies withing a <pb_type> structure, as it indicates the method of power estimation to be used for that block type.

Take a look at the k6_N10_mem32K_40nm.xml architecture for an example.  The full documentation is here: http://vtr.readthedocs.io/en/latest/vtr/power_estimation/

closed 2016-05-10 by jgoeders