verilog-to-routing/vtr-verilog-to-routing #130: 90nm Technology Properties File?

Ahmad90, 2016-05-10:
Hi,

i work on 90nm bulk technology file in Hspice, but now i cant create technology file whit script "<vtr>/vtr_flow/scripts/generate_cmos_tech_data.pl <tech_file> <tech_size> <vdd> <temp>" for VPR power.

i run the script in windows and Linux with Hspice environment! but output file not create. 

the error is:
could not find fallt in spice output.

Who can help me? or how has the 90nm-bulk technology file? 

tnx a lot...

jgoeders, 2016-05-10:
Based on the error, it is trying to run some hspice simulations on your technology, but the output is not as expected, and it can't find the fall time (fallt) of the inverter.  There could be many reasons for this.

First, I would try pulling down VTR from github and see if it works in the latest version.  There have been some fixes to the technology characterization scripts since the VTR 7.0 release.

If you are permitted to share your technology file with me, you can send it to me: jgoeders at ece.ubc.ca and I will see if the scripts work for me.

jgoeders, 2016-05-10:
OK, I tried performing characterization on the PTM 90nm file from here http://ptm.asu.edu/modelcard/2006/90nm_bulk.pm  

I ran into an error because the scripts require this folder to be present: vtr_flow/scripts/spice/temp

If you create this folder (or update to the lastest version on git, where I have added this folder), it should work.

I tested it using this command:
`./generate_cmos_tech_data.pl ../tech/PTM_90nm/90nm.pm 90e-9 1.0 85 > ../tech/PTM_90nm/90nm.xml &`

You can look in scripts/generate_tech_xml_files.sh for an example.  You will see I chose to use 1.0V for the voltage for this technology.  I'm not sure if that is correct - you might need to fix that.

Ahmad90, 2016-05-11:
tnx for your help. i cant run scrip!!!

my technology file in Hspice is attach.

VDD is 1.2V
Temp is 85

[90nm_bulk.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/258664/90nm_bulk.txt)

jgoeders, 2016-05-14:
[90nm.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/264532/90nm.zip)

Here is the xml file for your technology.  Let me know if you have any other questions.  As I mentioned you should probably use the repository version of VTR, as I think there are a few power related bug fixes since the VTR 7 release.

closed 2016-05-14 by jgoeders