verilog-to-routing/vtr-verilog-to-routing #131: Too many DSP blocks required

DriesVercruyce, 2016-06-07:
When VPR is used to pack the Titan benchmark designs on the Titan architecture, more DSP blocks are used than strictly required. For example neuron uses around 92 DSP blocks, while 89 should be sufficient. Benchmark des90 uses around 50 DSP blocks instead of 44.

I have modified the stratixiv_arch.timing.xml architecture and changed a function in cluster.c in an attempt to solve these problems

Firstly I added pack patterns to all modes in the DSP block. In the current version of the Titan architecture pack patterns are added to a few modes. By adding pack patterns to all modes, less DSP blocks are required and no increase in total packing runtime is obtained. Therefore I wonder why not all pack patterns are added in this architecture?

Secondly I suppose there is a bug in the get_free_molecule_with_most_ext_inputs_for_cluster function. In this function, the number of available inputs is counted by checking for all inputs if it is already used or not. Although, to my opinion, the number of used inputs is currently counted instead of the number of free inputs:
pack/cluster.c line 1128 => if (cur_pb->pb_stats->input_pins_used[i][j] **!=** OPEN) inputs_avail++;
Therefore I changed this line to:
if (cur_pb->pb_stats->input_pins_used[i][j] **==** OPEN) inputs_avail++;

With these modifications, it is possible to pack neuron and des90 with 89 and 44 DSP blocks respectively.

@EliasVansteenkiste

kmurray, 2016-06-07:
Thanks for the bug report! I've committed the pin counting fix for the clusterer.

I originally only added pack patterns for the most common DSP modes to stratixiv_arch.timing.xml due to time constraints.  It would be good to have patterns for all DSP modes.

If you can provide the stratixiv_arch.timing.xml with updated pack patterns, I will look at integrating them.

DriesVercruyce, 2016-06-08:
This is the architecture I used to pack the Titan benchmarks with all pack patterns.

[stratixiv_arch.timing.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/305649/stratixiv_arch.timing.txt)

kmurray, 2017-03-11:
I've added pack patterns for all DSP modes to the titan architecture capture.  They will appear in the next Titan release.

closed 2017-03-11 by kmurray