verilog-to-routing/vtr-verilog-to-routing #141: problem with vpr graphical window

PetrosGian, 2016-07-04:
This is not a bug I believe however I don't know how else to find answers. I tried reading the manual that you have posted but I cannot make, the graphic environment of VPR, work.

My OS is ubuntu 14.04
My VTR version is the current latest

The run_vtr_task.pl basic_flow worked as it was intended.
I have trouble enabling graphics as I don't quite understand what do I need to change. 
I tried going to the makefile of VPR and did the following:
ENABLE_GRAPHICS = true in vpr/Makefile, as mentioned in the documentation here : [http://vtr.readthedocs.io/en/latest/vpr/graphics/](url) 

However, when I am trying to execute a sample architecture in the vpr directory:
./vpr sample.arch.xml or1200.latern 
there is no graphical window poping up. 

Any help would be greatly appreciated. If I have asked in the wrong area, please let me know if there's somewhere else I could rather contact. 

Thank you in advance,
Petros

kmurray, 2016-07-06:
Hi Petros,

Once you have set `ENABLE_GRAPHICS = true` in the Makefile you will need to re-build VPR.

The easiest way to do this is:
`make clean`
followed by:
`make`
in the root of the VTR source tree.

`make` may prompt you to install some required libraries if your system does not have them installed.

Once VPR has been re-built it should automatically pop-up the GUI window when it is run.

PetrosGian, 2016-07-06:
Thank you for the response.
I managed to make it work yesterday.
I was actually doing 'make' without the 'make clean'.

closed 2016-07-06 by PetrosGian