verilog-to-routing/vtr-verilog-to-routing #144: run_vtr_flow.pl -starting_stage ABC

werneazc, 2016-07-06:
Hi,

I run a benchmarking with **run_vtr_flow.pl** and started the script at abc. If I do so, the used architecture file is mentioned with the name of the circuit. For parsing I need the name of the architecture in _vpr.out_ or _vpr_stdout.log_.

It would be great if one can fix this naming error.

example:
Architecture file: mycircuit
Circuit name: mycircuit.blif

Best regards
André

kmurray, 2016-07-06:
Hi André,

What is the exact command you are running?

I tried (where `<vtr>` is the root of the VTR source tree):
```
<vtr>/vtr_flow/scripts/run_vtr_flow.pl <vtr>/vtr_flow/benchmarks/blif/6/alu4.blif <vtr>/vtr_flow/arch/timing/k6_frac_N10_40nm.xml -starting_stage abc
```

Which produced:
```
Architecture file: k6_frac_N10_40nm.xml
Circuit name: alu4.blif
```
As expected in both `vpr.out` and `vpr_stdout.log`.

werneazc, 2016-07-06:
Hey, I used the following command:

VTR_ROOT/vtr_flow/scripts/run_vtr_flow.pl mycircuit.blif myarch.xml -starting_stage abc -temp_dir ./myarch

VTR_ROOT is the path to my directory of vtr git repository.
I executed it not in the VTR source tree.

Best regards
André

kmurray, 2016-07-06:
This should now be fixed by 56cdc03dd.

The bug was caused by incorrect handling of relative paths without any `/`s in them.  The updated code should be more robust.

closed 2016-07-06 by kmurray