verilog-to-routing/vtr-verilog-to-routing #146: Improve BLIF Parser Error Reporting

kmurray, 2016-07-11:
From Dave Evans:
> I've had a few issues with the blif parser like this where it just kind of breaks with an unhelpful assertion or segfault. In the future would it be possible to get some helpful error messages incorporated into the parser? It seems like many of these things would be easy to detect
>
> Some of the things I've run into:
> 1. Top level module must be the first module
> 2. subckt and architecture file have mismatching ports
> 3. Requirement for unconn nets with default modules (multiply, adder, etc)

I agree that (1) and (2) are problems for which we should detect and give friendlier error messages.

I also agree that using 'unconn' nets is akward -- but I'm not sure there is an alternative, since BLIF does not really support the idea of ports which are unconnected.

DFarce, 2016-07-11:
I've noticed that some tools simply do not define unconnected ports. For example an adder cell could be defined as:

> `.subckt adder A[0]=my_num_1 B[0]=my_num_2 sumout[0]=my_output`

Versus the way ODIN_II handles this case:
> `.subckt adder A[0]=my_num_1 B[0]=my_num_2 cin[0]=unconn sumout[0]=my_output` cout[0]=DUMMY_COUT[0]

However, it's not really something where there is a "correct" way, so it's understandable to keep things the same.

kmurray, 2016-11-04:
I've re-written the BLIF parser, so the latest trunk code (tested with d17170f) should address these issues.

1. VPR no longer requires that the top level is the first `.model` in the BLIF file.  

 Note that VPR still requires a flat netlist; only one `.model` can contain logic, all other `.model`s must be black boxes (`.blackbox`).

2. Mismatches between a subckt and the architecture now generate errors.
 For instance if the `.model` is missing in the architecture file:
 ```
 Error 1: 
 Type: Blif file
 File: /project/trees/vtr/vtr_flow/benchmarks/vtr_benchmarks_blif/diffeq1.blif
 Line: 296
 Message: Failed to find matching architecture model for 'multiply'
 ```

 Or if there is a port mismatch:

 ```
 Error 1: 
 Type: Blif file
 File: ch_intrinsics.pre-vpr.blif
 Line: 316
 Message: Found no matching port 'invalid_port' on architecture model 'single_port_ram'
 ```

 Or if the bit index on the port is out of range:
 ```
 Error 1: 
 Type: Blif file
 File: ch_intrinsics.pre-vpr.blif
 Line: 316
 Message: Port 'addr[15]' on architecture model 'single_port_ram' exceeds port width (15 bits)
 ```

3. Unspecified `.subckt` ports are now assumed to be disconnected.

 Note that VPR still supports the `unconn` net, which is needed to handle disconnected inputs to `.names`.

 `.subckt` pins can be specified either by:
 - connecting them to the `unconn` net (the traditional VPR method), or by
 - leaving them unspecified

 As a result these two `.subckt` instantiations are now equivalent:
 ```
.subckt single_port_ram \
clk=top^clk \
data=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~546 \
addr[0]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~541 \
addr[1]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~542 \
addr[2]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~543 \
addr[3]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~544 \
addr[4]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~545 \
addr[5]=unconn \
addr[6]=unconn \
addr[7]=unconn \
addr[8]=unconn \
addr[9]=unconn \
addr[10]=unconn \
addr[11]=unconn \
addr[12]=unconn \
addr[13]=unconn \
addr[14]=unconn \
we=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~554 \
out=top.memory_controller+memtroll.single_port_ram+str^out~0
 ```

 ```
.subckt single_port_ram \
clk=top^clk \
data=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~546 \
addr[0]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~541 \
addr[1]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~542 \
addr[2]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~543 \
addr[3]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~544 \
addr[4]=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~545 \
we=top.memory_controller+memtroll^MULTI_PORT_MUX~8^MUX_2~554 \
out=top.memory_controller+memtroll.single_port_ram+str^out~0
 ```

closed 2016-11-04 by kmurray