verilog-to-routing/vtr-verilog-to-routing #147: Odin memory errors

jean-w, 2016-07-13:
I came across some memory errors in Odin while running Valgrind with these circuits (not really an exhaustive list, just what I happened to test): 

- vtr_flow/benchmarks/verilog/mkSMAdapter4B.v
- vtr_flow/benchmarks/verilog/or1200.v
- vtr_flow/benchmarks/verilog/raygentop.v

The command to reproduce the memory errors is :   
`    valgrind ./odin_II.exe -V ../vtr_flow/benchmarks/verilog/raygentop.v`

I've also attached the valgrind output [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/361734/valgrindout.txt)

PanosNB, 2016-07-13:
Thanks for spotting this @jean-w 

I pushed a fix that solves these issues (554d207d8a35dd61c4b504659b133f4b2584ed4f)

You can close the issue, if you don't find anything else on Odin

By the way, you can do a "git rebase" to avoid the extra merging commits after you pull

jean-w, 2016-07-13:
Great. Thank you!

closed 2016-07-13 by jean-w