verilog-to-routing/vtr-verilog-to-routing #149: VPR assigns tracks incorrectly

eshellko, 2016-07-18:
I am using Wilton switchbox which has Fs = 3.
Function 'load_sblock_pattern_lookup' convert it into Fs_per_side parameter equal to 1 for my case:
```
	int Fs_per_side = 1;
	if (Fs != -1) {
		Fs_per_side = Fs / 3;
	}
```
During construction of sblock opposite side multiplexer distribution uses following formula:
```
						int mux = ((side_ccw_incoming_wire_count + side_cw_incoming_wire_count)
								* Fs_per_side + opp_incoming_wire_count	* (Fs_per_side - 1))
								% num_wire_muxes[to_side];
```
Where iterator 'opp_incoming_wire_count' ignored as multiplied-by-0. This cause many tracks connected to same multiplexer (defined by 'mux') which produce 43-to-1 MUX for my design with channel width equal to 100 and segment length equal to 4.
As main reason for this function is to create
> perfectly balanced and M-to-N assignment

my guess is that for architectures with Fs=3 another equation like 'opp_incoming_wire_count	* Fs_per_side' should be used. Can anyone approve or reject my guess?

kmurray, 2016-07-18:
I'm not an expert on the RR graph generator so I can't really comment.

@opetelin Any thoughts on this?

opetelin, 2016-07-18:
Hey,

It's been a long time since I looked at this code, but as mentioned the purpose is to assign mux connections such that, on any given side, mux size is imbalanced by at most 1.

Looking at the graphics from VPR I can not spot any imbalance beyond 1 on any core/corner/edge routing muxes that I checked.

However, a 43-to-1 mux in a W=100 L=4 architecture doesn't sound correct, unless the connection block Fc_out values are high. Can you post an example of this happening?

opetelin, 2016-07-18:
I took a second look at the code in question, and while it does look a bit sketchy it appears to work fine. I will add a comment in the code to clarify the function.

The short of it is that in the VPR 7.0 switch block code (not to confuse it with the custom switch block description language I added more recently) wire segments that pass through a switch block have an implicit connection to the opposite side which counts towards the Fs (therefore at midpoints wire segments connect only to the two adjacent sides).

The code in question appears to incorrectly assign the same destination mux to all the wires segment midpoints connecting to the opposite switch block side. However, this assignment is never actually used. Later in the code (build_rr_chan() in rr_graph.c), the rr graph generator explicitly skips connections from wire segment midpoints to the opposite switch block side.

However, there does appear to be a mux balancing issue when Fs > 3 is used. I would suggest using the custom switch block descriptions for these kinds of architectures (https://docs.verilogtorouting.org/en/latest/arch/reference/#custom-switch-blocks).

closed 2016-07-20 by kmurray