verilog-to-routing/vtr-verilog-to-routing #150: incorrect results when using a gate-level verilog netlist in VTR flow

berbagci, 2016-07-18:
Hi,

I have a sample verilog code (i.e., sha in standard MCNC benchmarks) and mapping it to a gate level netlist by using only a limited subset of gates
[ISSUE.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/369759/ISSUE.zip)

 (2input and/or, inverters, and flops). Then feeding this gate-level verilog into vtr flow. However, the logic doesn't seem to be translated properly (i'm seeing input output ports, and some constant gens and a small logic). Odin step seems to be fine, by fine I mean I'm seeing some logic mapped in truth table formats but this seems to be lost after abc step. 
Files (fpga architecture and verilog) attached to reproduce the issue. Also using the latest version in the trunk.

Any help is appreciated.
Thanks

kmurray, 2016-07-19:
I agree that most of the logic appears to be swept away by ABC.

However running logical equivalence (e.g. `abc -c "sec sha_gate_level.odin.blif sha_gate_level.abc.blif"`) shows that the Odin and ABC output netlists are equivalent.  This implies that ABC synthesis is working correctly and simply synthesizing away un-needed logic.  

Perhaps there is an issue with the gate-level verilog netlist?

berbagci, 2016-07-20:
hmm, that's really weird. let me check the gate-level verilog netlist.
Thank you

closed 2016-07-20 by kmurray

berbagci, 2016-07-21:
also, thank you so much for your quick responses. I really appreciate it.
thanks again!