verilog-to-routing/vtr-verilog-to-routing #151: using  previos place and route files

berbagci, 2016-07-19:
Hi,

I have an arch file and used VTR flow to get delay values of various benchmarks on this arch file. And I made a copy of the same arch file. The only difference is the delay values for different blocks (LUT delays, etc.). I want to get the delay values for the same benchmarks on this copied architecture with different delay values for FPGA blocks. Hence, I do want to use the exact same placement and routing (assuming the architecture parameters didn't change, which is the case). Is that possible?  

Thanks

kmurray, 2016-07-19:
Yes this is possible, however be aware that the result _will not be a fair comparison_ - since the placement/packing/routing will not be optimized for the new delay values.

You can pass VPR the `--route` option which will tell it to perform only routing. VPR will then attempt to read the packing (.net) and placement (.place) files instead of re-doing those stages.

closed 2016-07-19 by kmurray

berbagci, 2016-07-19:
Thank you for your response. 
That's fine, I'm basically trying to find delay values for different supply voltages (since only the delay values should change when the supply voltage on a programmed FPGA changes).
Thanks again

berbagci, 2016-07-20:
so I'm running vpr using the following command:

./vpr ../vtr_flow/tasks/basic_flow/original/secure.xml/aes_cipher_top_all.v/secure.xml aes_cipher_top_all --route --blif_file "../vtr_flow/tasks/basic_flow/original/secure.xml/aes_cipher_top_all.v/aes_cipher_top_all.pre-vpr.blif" --net_file "../vtr_flow/tasks/basic_flow/original/secure.xml/aes_cipher_top_all.v/aes_cipher_top_all.net" --place_file "../vtr_flow/tasks/basic_flow/original/secure.xml/aes_cipher_top_all.v/aes_cipher_top_all.place" --route_chan_width 120

but getting the following error message

Message: '../vtr_flow/tasks/basic_flow/original/secure.xml/aes_cipher_top_all.v/aes_cipher_top_all.place' - Architecture file that generated placement (aes_cipher_top_all.net) does not match current architecture file (../vtr_flow/tasks/basic_flow/original/secure.xml/aes_cipher_top_all.v/aes_cipher_top_all.net).

Although I'm using the exact same architecture as I did before. And this is not even with different delay values. I was just trying to run vpr with only route option. What am I missing? And is it possible to do this in vtr flow? 

Thanks for your help

berbagci, 2016-07-20:
Nvm, I think it works now, thanks again!

berbagci, 2016-07-20:
Actually, routing seems to be changed, I just want to recalculate the delay for the same architecture expect for delay values. Does VPR support this? Thanks

kmurray, 2016-07-20:
No, currently this is not supported by VPR.  This would be a useful feature so I've filled issue #154 to add support for this