verilog-to-routing/vtr-verilog-to-routing #157: LUT input pin swapability

berbagci, 2016-09-16:
So VPR reorders LUT inputs, is there a way to turn this off? I'm particularly interested in turning this feature off for one of the inputs, the input that can be directly connected to FF (input selection mux from one of the LUT inputs to FF input). This is supported by the generic architecture described in original VPR paper. 

Let's say LUT input C is also connected to FF input (FF input selection mux).
The problem is when inputs are reordered, the input for FF (in some cases) seems to be connected to some other LUT input (e.g. D) which in fact has no connection.

So I would like to prevent VPR from doing that since this is not described in the arch file.

kmurray, 2016-09-19:
This sounds like a bug.

Do you have an example circuit and architecture which exhibit the issue?

berbagci, 2016-09-19:
I'm attaching the files.
The folder has both an example circuit and architecture. 
Thanks
[dua_sample.v.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/481088/dua_sample.v.zip)

berbagci, 2016-09-19:
Sorry, the previous folder has all the intermediate files from vtr task as well
this one has only the circuit and arch.

[example.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/481092/example.zip)

berbagci, 2016-10-03:
so could you replicate the issue? thanks

closed 2025-05-06 by github-actions[bot]