verilog-to-routing/vtr-verilog-to-routing #160: Too many M9K block required

DriesVercruyce, 2016-10-12:
I have a problem with packing memory slices when the Titan architecture is used (attached in _files.zip_). In the special case described below, too many M9K blocks are required because only one memory slice is packed in some of them. I encountered this problem in the sparcT2_core benchmark design.

In [files.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/523961/files.zip) I attached four blif files. All these blif files contain a set of stratixiv_ram_block.opmode{dual_port}.output_type{comb}.port_a_address_width{7}.port_b_address_width{7} memory slices. All these memory slices have the same address and clock signals. The _portawe_ and _ena0_ input control signals are connected to a net and all other control signals are unconnected. The script to generate these blif files is also attached in _files.zip_.

In the first two blif files (_example_one_signal_128_128.blif_ and _example_one_signal_129_129.blif_) no problem occurs. In these blif files, the _portawe_ and _ena0_ input control signals of all memory slices are connected to the same net (_s1_). The first blif file has 128+128=256 memory slices and the second one has 129+129=258 memory slices. As expected, these slices are packed into 8 M9K memory blocks, because each block can pack 36 slices.

In the other two blif files (_example_two_signal_128_128.blif_ and _example_two_signal_129_129.blif_) a problem occurs during packing. In these blif files there are two different nets (_s1_ and _s2_) that drive the input control signals _portawe_ and _ena0_. Half of the slices have signal _s1_ at the input of _portawe_ and _ena0_, the other half has signal _s2_ at the input of _portawe_ and _ena0_. 
All memory slices in _example_two_signal_128_128.blif_ are packed into 8 M9K memory blocks, as espected. But when we add one more memory slice to both sets (in this case each set has 129 slices), then a problem occurs. We expect that these slices can be packied into 8 M9K blocks, but instead 20 M9K blocks are required (_example_two_signal_129_129.blif_).

I am currently using this version of VTR:
Version: Version 7.0.7
Revision: 75b47d3-dirty

@EliasVansteenkiste

kmurray, 2016-10-12:
Thanks for the bug report Dries.

I haven't reproduced it myself, but I can thing of two possible causes:

1. There is a limitation in the Titan architecture (e.g. available pins/connectivity) which prevents the slices from packing together,
2. Or this case is exposing bad behavior in the packing algorithm (i.e. the packer refuses to pack them together even though it would be legal).

If the issue is (1), we would need to check if this is legal in the Stratix IV architecture, and if it is update the Titan model appropriately.
If the issue is (2), then we would need to figure out *why* the algorithm is failing to handle this case correctly.

Hopefully I'll have some time to take a look at this in the near future.

kmurray, 2016-10-28:
I've been able to reproduce the issue with the latest version. After some investigation I've concluded that the root cause is (2). Since the packer is a heuristic algorithm it is not guaranteed to find the best packing, so I believe the root cause is simply sub-optimality in the packing algorithm.

On the problematic circuit you provided (i.e. example_two_signal_129_129.blif) I found that small tweaks to the algorithmic parameters (e.g. setting AAPACK_MAX_NET_SINKS_IGNORE to 512, or re-ordering seed blocks) could cause the problematic circuit to pack into 8 (instead of 20) M9Ks.  

However I don't believe such tweaks fundamentally fix the problem - they simply perturb the algorithm (analogous to changing the seed for placement).

It is possible that the packing algorithm could be modified to more gracefully handle such cases, but it is not immediately obvious to me what changes should be made (perhaps by treating RAMs as a special case?).

closed 2017-05-31 by kmurray