verilog-to-routing/vtr-verilog-to-routing #161: VTR Installation issue : vtr_math.cpp

akrish21, 2016-10-16:
I am seeing the following erros while "make" from VTR root

#1 
Users/akrish/Applications/vtr-verilog-to-routing/libvtrutil/src/vtr_error.h:25:21: error: implicit instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >'
        std::string filename_;
                    ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/iosfwd:193:33: note: template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
In file included from /Users/akrish/Applications/vtr-verilog-to-routing/libvtrutil/src/vtr_math.cpp:4:

#2 
Users/akrish/Applications/vtr-verilog-to-routing/libvtrutil/src/vtr_math.cpp:33:15: error: no matching constructor for initialization of 'vtr::VtrError'
        throw VtrError("linear_interpolate_or_extrapolate: cannot interpolate/extrapolate based on less than 2 (x,y) pairs", __FILE__, __LINE__); 

#### Steps to Reproduce (for bugs)
1.  make
2. 
3. 
4. 

#### Your Environment
* VTR revision used: VTR 7.0
* Operating System and version: Mac OS Sierra 10.12
* Compiler version:
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.38)
Target: x86_64-apple-darwin16.0.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

Any help regarding this please ?

Thanks,
AK

kmurray, 2016-10-17:
This was caused by a missing header file, and should be fixed in 4623695.

Interestingly this caused no problem with GCC on Ubuntu.

We don't actively test with LLVM on Mac, so you may encounter other issues.  If so, please let us know.

akrish21, 2016-10-17:
Thanks for the fix for the above fails. I am now seeing the following after pulling your latest update. 

Error as follows:

#####################
[  2%] Building CXX object libvtrutil/CMakeFiles/libvtrutil.dir/src/vtr_util.cpp.o
/Users/akrish/Applications/vtr-verilog-to-routing/libvtrutil/src/vtr_util.cpp:88:42: warning: format string is not a string literal [-Wformat-nonliteral]
    int len = std::vsnprintf(nullptr, 0, fmt, va_args_copy); 
                                         ^~~
/Users/akrish/Applications/vtr-verilog-to-routing/libvtrutil/src/vtr_util.cpp:102:47: warning: format string is not a string literal [-Wformat-nonliteral]
    len = std::vsnprintf(buf.get(), buf_size, fmt, args);
                                              ^~~
/Users/akrish/Applications/vtr-verilog-to-routing/libvtrutil/src/vtr_util.cpp:204:103: error: use of undeclared identifier 'errno'
        throw VtrError(string_fmt("Error opening file %s for %s access: %s.\n", fname, flag, strerror(errno)), __FILE__, __LINE__);        
                                                                                                      ^
2 warnings and 1 error generated.
####################################
Can you take a look at this as well ?

Thanks,
AK

kmurray, 2016-10-17:
Also another missing header, which should be fixed in c893656.

I've re-named this issue to reflect the general problem, and I'll leave it open for any more problems.

akrish21, 2016-10-17:
alrighty, sounds good.

Here is the next one. Let me know if there is another intelligent way to keep collecting these errors, but step through the make process till end so you can fix everything once :)

clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated
/Users/akrish/Applications/vtr-verilog-to-routing/libarchfpga/echo_arch.c:400:2: error: use of undeclared identifier 'free'
        free(tabs);
        ^
1 error generated.

kmurray, 2016-10-17:
The free issue should be fixed with 0f128de.

I don't have access to Mac to test on, so I'm not sure there is a better way.

akrish21, 2016-10-18:
Alrighty. No  Problem. Here are next set of errors I saw -

/Users/akrish/Applications/vtr-verilog-to-routing/vpr/SRC/base/place_and_route.c:616:37: error: use of undeclared identifier 'floor'
                        chan_width.y_list[i + 1] = (int) floor(cfactor * comp_width(&chan_y_dist, x, separation) + 0.5);
                                                         ^
/Users/akrish/Applications/vtr-verilog-to-routing/vpr/SRC/base/place_and_route.c:665:22: error: use of undeclared identifier 'exp'
                val = chan->peak * exp(-val);
                                   ^
/Users/akrish/Applications/vtr-verilog-to-routing/vpr/SRC/base/place_and_route.c:670:17: error: use of undeclared identifier 'fabs'
                val = (float) fabs((double) (x - chan->xpeak));

Should be some missing math header or something, but im really not sure. 

Thanks,
AK

kmurray, 2016-10-18:
Fixed in 1772b1c.

It seems that Ubuntu implicitly includes most of these headers in other header files, while Mac does not. So the fix is always fairly straight forward, we just need include the appropriate header file.

akrish21, 2016-10-18:
Well that is very unfortunate for ppl like me who depend on mac to do some opensource work in our industry.. Anyways, here is the next one - 

/Users/akrish/Applications/vtr-verilog-to-routing/vpr/SRC/base/vpr_api.c:278:36: error: use of undeclared identifier 'sqrt'
    int current = vtr::nint((float)sqrt((float)num_blocks)); /* current is the value of the smaller side of the FPGA */

kmurray, 2016-10-18:
Fixed in 2276288.

kmurray, 2017-04-10:
We now actively test building VTR with GCC 4.9/5/6 and LLVM 3.6/3.8 on linux (see [buildbot](http://builds.verilogtorouting.org:8080/builders/build_configs)).  Hopefully this will prevent these types of issues from creeping into the code base in the future.

closed 2017-04-10 by kmurray