verilog-to-routing/vtr-verilog-to-routing #162: An issue about the switch box permutation rules

CenIII, 2016-10-19:
I think the switch permutation building process have something wrong ( vtr-verilog-to-routing/vpr/SRC/route/build_switchblocks.c ). When we define a permutation rule in architecture description file like below: (it's from part of the arch description file "k4_N8_topology-0.85sL2-0.15gL4-on-cb-off-sb_22nm_22nm.xml")

<switchblocklist>
		<switchblock name="wilton_turn_core" type="unidir">
			<switchblock_location type="CORE"/>
			<switchfuncs>
				<func type="lt" formula="W-t"/>
				<func type="lb" formula="t-1"/>
				<func type="rt" formula="t-1"/>
				<func type="br" formula="W-t-2"/>
				<func type="tl" formula="W-t"/>
				<func type="bl" formula="t+1"/>
				<func type="tr" formula="t+1"/>
				<func type="rb" formula="W-t-2"/>
			</switchfuncs>
			<wireconn from_type="l2s" to_type="l2s" from_switchpoint="0" to_switchpoint="0"/>
			<wireconn from_type="l2s" to_type="l2s" from_switchpoint="1" to_switchpoint="0"/>
			<wireconn from_type="l4g" to_type="l4g" from_switchpoint="0" to_switchpoint="0"/>
			<wireconn from_type="l4g" to_type="l2s" from_switchpoint="0" to_switchpoint="0"/>
		</switchblock>

		<switchblock name="wilton_turn_perimeter" type="unidir">
			<switchblock_location type="PERIMETER"/>
			<switchfuncs>
				<func type="lt" formula="W-t"/>
				<func type="lb" formula="t-1"/>
				<func type="rt" formula="t-1"/>
				<func type="br" formula="W-t-2"/>
				<func type="tl" formula="W-t"/>
				<func type="bl" formula="t+1"/>
				<func type="tr" formula="t+1"/>
				<func type="rb" formula="W-t-2"/>
			</switchfuncs>
			<wireconn from_type="l2s" to_type="l2s" from_switchpoint="0,1" to_switchpoint="0"/>
			<wireconn from_type="l4g" to_type="l4g" from_switchpoint="0" to_switchpoint="0"/>
			<wireconn from_type="l4g" to_type="l2s" from_switchpoint="0" to_switchpoint="0"/>
		</switchblock>

		<switchblock name="wilton_straight" type="unidir">
			<switchblock_location type="EVERYWHERE"/>
			<switchfuncs>
				<func type="lr" formula="t"/>
				<func type="bt" formula="t"/>
				<func type="rl" formula="t"/>
				<func type="tb" formula="t"/>
			</switchfuncs>
			<wireconn from_type="l2s" to_type="l2s" from_switchpoint="0" to_switchpoint="0"/>
			<wireconn from_type="l4g" to_type="l4g" from_switchpoint="0" to_switchpoint="0"/>
			<wireconn from_type="l4g" to_type="l2s" from_switchpoint="0" to_switchpoint="0"/>
		</switchblock>
	</switchblocklist>

With the permutation definition above, and the building rule according to the current "build_switchblocks.c", we can't build a switch block in which Fs=3. The result is, the connections from "l4g" will always be 6 (2 for each opposite side), rather than 3 (1 for each opposite side), because there's a "for" loop going through each "wireconn" tag and made the permutation defined above, and since "l4g" appears twice as "from_type", the wire connections from each "l4g" segment is 2x3=6. 

You may argue that that's what it is should be like. But the question is, with such a building rule, how could we define a switch box, in which we require that each segment from one side (whether it's "l4g" or "l2s") should connects to 3 opposites (1 for each opposite), **and** among the 3 opposites a "l4g" segment connects to, **"l2s" and "l4g" are both allowed**?

CenIII, 2016-10-19:
By the way the pull request I made several days ago was good. Why not consider merging that? ;)

opetelin, 2016-10-20:
Hi Cenlll,

I'm the author of the switch block code; I graduated last month, so my responses are gonna be slow.

The switch block pattern produced is indeed as it should be. We see that in each <switchblock> definition, the specified permutation functions are applied once to the l4g-->l2s connections, and once to the l4g-->l4g connections (in accordance with the wireconn entires). So for each side that an l4g wire segment connects to, it will connect once to an l4g wire and once to an l2s wire, for a total flexibility of fs = 2x3 = 6 as you noted.

If you wanted to strictly maintain a switch block flexibility of fs=3 for the l4g wires, you could instead specify a wireconn entry like this:

`<wireconn from_type="l4g" to_type="l2s, l4g" from_switchpoint="0" to_switchpoint="0"/>`

With the to_type="l2s, l4g" entry, the legal set of destination wires to which we can connect will be all l2s and l4g wire segments that have a switchpoint of 0 on the switch block side in question. The source l4g wire will then connect to either an l4g or an l2s wire in accordance with the permutation function, maintaining an overall switch block flexibility of fs=3 for l4g wires. So the key thing here is that the permutation functions are applied to each "wireconn" entry specified.

However, as a side note, maintaining a flexibility of fs=3 for the l4g wires is precisely what we *didn't* want to do with this architecture because it doesn't allow us to maintain a well-defined hierarchy between the l4g and l2s wire types -- interconnect design is much simpler if we know for a fact that connections between wire types are consistent at each switch block in the FPGA.

To summarize: seems to me that the switch block code is working as intended. An fs=3 can be enforced for the l4g wires by modifying the switch block description.

I haven't looked at your pull request in detail, but I'll check it out soon.

Oleg

opetelin, 2016-10-20:
Also, if you haven't yet, check out the switch block format documentation:

https://docs.verilogtorouting.org/en/latest/arch/reference/#custom-switch-blocks

CenIII, 2016-10-21:
Thank you for your reply. Yes I can understand now. One thing I didn't realize was that the wireconn entry can be written like that:

<wireconn from_type="l4g" to_type="l2s, l4g" from_switchpoint="0" to_switchpoint="0"/>

That's great. Now I can see that with such flexibility, basically all kinds of connection patterns can be obtained. 👍 👍 👍

closed 2016-10-21 by opetelin