verilog-to-routing/vtr-verilog-to-routing #164: Cannot "make" vtr release on Ubuntu 14.04 32-bit OS

g2-aj, 2016-11-12:
##### Installation and Make issue:
I downloaded vtr release 7.0, and after unpacking ran "make". It said I didn't have all the necessary packages so I ran "make packages".
It installs the necessary packages (ctags, bison, flex, g++) and then stops with the following:
"""
make[1]: *** No rule to make target `$(dir)', needed by `OBJ/get.o'.  Stop.
make: *** [pcre] Error 2
"""

What could I be missing?

#### Environment
* VTR revision used: Downloaded release 7.0 from http://www.eecg.utoronto.ca/vtr/download.allow.html
* Operating System and version: Ubuntu 14.04 - 32 bit
With 
GNU Make 3.80
Copyright (C) 2002  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

kmurray, 2016-11-14:
Have you tried performing `make clean` and then `make`?

g2-aj, 2016-11-14:
Yes I tried that, I get the same error as above.

kmurray, 2016-11-14:
I was able to install the release with the following on Ubuntu 14.04 32 bit:
```
$ tar -xzvf vtr_release_7.0_full.tar.gz
$ cd vtr_release
$ make packages
$ make
```

Can you provide the full output of the `make clean` and `make` commands?

g2-aj, 2016-11-15:
Hi,

Okay I reinstalled my VM and tried it again and this time it all worked perfectly. So I'm not sure what I had corrupted in the previous install, but anyways all is good. Thanks for getting back to me!

closed 2016-11-15 by g2-aj