verilog-to-routing/vtr-verilog-to-routing #165: VPR packing timing metrics in Titan article

DriesVercruyce, 2016-11-16:
I am working on a new packing algorithm. To analyse the quality of results, we want to compare the new packing algorithm with the built-in packing algorithm in VPR. As benchmarking test we use the Titan benchmark suite. 

In my opinion, there are three closely related metrics to analyse the post-routing speed-performance of the circuit:
1) 'Critical path delay'
2) 'Geometric mean intra-domain period'
3) 'Fanout-weighted geomean intra-domain period'
Each of these metrics is reported by VPR after routing. We wonder which of these metrics is used in Table IX in the article ‘Timing Driven Titan: Enabling Large Benchmarks and Exploring the Gap Between Academic and Commercial CAD’ ? We think that the 'critical path delay' is used for single-clock circuits and the 'geometric mean intra-domain period' is used for multi-clock circuits. Is that correct?

In the article the following is stated in section 7.1: "The benchmark I/Os were constrained to a virtual I/O clock with loose input/output delay constraints. Paths between netlist clock-domains and the I/O domain were analyzed, to ensure that the tools can not (unrealistically) ignore I/O timing"
Although, one line above the following is written: "All paths crossing netlist clock-domains were cut, ensuring that the tools can focus on optimizing each clock independently."

If I understand this correctly, the paths between netlist clock-domains and the I/O domain were analyzed and taken into account to find the critical path delay but are not optimized during packing (because all paths crossing netlist clock-domains are cut).

In section 7.2 the following is written: "For multi-clock circuits we report the geometric mean of critical path delays across all clocks, excluding the virtual I/O clock."

Does this mean that all Titan benchmark designs are multi-clock circuits because all designs have a clk and a virtual_io_clock? If that is not the case, then single and multi-clock circuits are treated differently. Or did I misunderstood this in the Titan article?

@EliasVansteenkiste

kmurray, 2016-11-17:
There are basically three types of paths to consider:

1. Register-to-Register paths where both registers are clocked by the *same* netlist clock domain (i.e. intra-domain transfers)
2. Register-to-Register paths where each register is clocked by *different* netlist clock domains (i.e. inter-domain transfers)
3. Input-to-Register and Register-to-Output paths where the I/O launch/capture clock is called a 'virtual' clock since it does not appear in the netlist

For the results in the Titan paper we specified the following:

- VPR should analyze paths of type (1)
- VPR should not analyze (and hence not optimize) paths of type (2)
- VPR should analyze paths of type (3) to consider I/O timing

We then reported the geometric mean critical path delay (CPD) across all intra-clock domain transfers, i.e. paths of type (1).

So to clarify, by "multi-clock circuits" we mean circuits with multiple netlist clocks (i.e. we do not consider I/O paths of type (3) when calculating the goemean CPD). 
So the reported geomean CPD is the same for single clock circuits (i.e. the geomean of the single CPD) and multi-clock circuits (i.e. the goemean of all intra-domain CPDs).

Hope this helps.

kmurray, 2017-01-02:
I should also point out that the SDC files used are included in the Titan release, so you can see how the timing constraints are configured.

For example the SDC file used by VPR for the `neuron` benchmark is located at `benchmarks/titan23/neuron/sdc/vpr.sdc` in the Titan release.

closed 2017-01-02 by kmurray