verilog-to-routing/vtr-verilog-to-routing #166: Possibility of creation of "FPGA" without using LUT

citylife4, 2016-12-08:
Is there any possibility to use VTR to create a netlist for a FPGA without using LUT's in a CLBs?
I'm trying to use ULB's ( http://ieeexplore.ieee.org/document/4569539/ ) on a small FPGA.

Follows a first scratch of a XML for a simple FPGA:

[soft_fpu_arch_timing.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/639862/soft_fpu_arch_timing.txt)

and this is the blif module for a ULB:

[ulb.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/639864/ulb.txt)

And using the VTR flow, the following error is produced:
[image]

Thanks!!

kmurray, 2016-12-08:
Yes this is possible.  

It looks like you are hitting a built in assumption in one of the scripts which is assuming a LUT-based architecture. However, the underlying tools (i.e. VPR) can target non-LUT based architectures.  So as a work around for the script issue you can [call VPR directly](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/).

The general approach is to treat the primitive logic type as a blackbox (.subckt in BLIF).

In your architecture you can then define 'sites' for the primitive within your CLB:
```xml
<pb_type name="clb">
  ...
  <pb_type name="ulb" blif_model=".subckt ulb" num_pb="1">
    <input name="in" num_pins="5"/>
    <output name="out" num_pins="2"/>
  </pb_type>
  ...
</pb_type>
```

To implement a circuit you will need to technology map your circuit to the logic primitive (e.g. `ulb`s) resulting in a BLIF netlist somewhat like:
```
 #'top' is the main netlist
.model top
.inputs a b c d e
.outputs fg

 #An instance of the ulb primitive
.subckt ulb in[0]=a in[1]=b in[2]=c in[3]=d in[4]=e out[0]=f out[1]=g

.end

 #We need to define what a subckt's external connectivity looks like
.model ulb
.inputs in[0] in[1] in[2] in[3] in[4]
.outputs out[0] out[1]
.blackbox
.end
```

You should then be able to run VPR directly to place and route the circuit onto your architecture:
```
vpr my_ulb_arch.xml my_ulb_netlist.blif
```

citylife4, 2016-12-08:
Thank you for the fast response.

But in the end if I run the command: 

`vpr my_ulb_arch.xml my_ulb_netlist.blif`

It will give me a netlist of "my_ulb_netlist.blif" for "my_ulb_arch" , but what I want to try, is to create a netlist of another circuit (in my example a counter) to be placed to my architecture that uses ULBs and not LUTs. Or the command was an example and I can use vpr to place and route any circuit.
For my example:

`vpr my_ulb_arch.xml counter.blif`

Thank you again.

kmurray, 2016-12-08:
To clarify, the arguments to VPR were place holders for what ever filenames you wish to use.  

So if you wanted to target an architecture described in the file 'ulb_arch.xml' and you want to map a circuit named 'counter.blif' onto that architecture you would use:
```
vpr ulb_arch.xml counter.blif
```

Hopefully this helps clarify.  For more details on VPR see the [documentation](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/).

closed 2016-12-10 by citylife4