verilog-to-routing/vtr-verilog-to-routing #167: VPR7 wrt VPR6 switch box internal wilton connection

lvezier, 2016-12-14:
Switch box connection seems to be wrong on peripheral switch box.
You will see in the following screenshots the switch box representation with VPR6 and with VPR7, for the same FPGA architecture xml file ( [flu.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/651930/flu.zip) ) .

#### Expected Behavior
[image]

#### Current Behavior
[image]

#### Context
Everything worked fine with VPR6.
Currently trying to upgrade to VPR7...
Using following xml architecture file : 
[flu.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/651930/flu.zip)

#### Your Environment
* VTR revision used: VPR 7.0.7
* Operating System and version: RHEL 6
* Compiler version:

kmurray, 2016-12-15:
I can not seem to reproduce the difference on my end.  Can you provide more information about how you are running VPR to produce the differences?

When I run your architecture through the VPR 6 release, VPR 7 release and the latest trunk code I get the same periphery switch blocks:

VPR 6:
[image]

VPR 7:
[image]

Latest Trunk:
[image]

lvezier, 2016-12-15:
Hi
Thanks for your answer
Perhaps my VPR 6 version is older than yours...
Where can I download your VPR6 version ?

lvezier, 2016-12-15:
Here is my launch option:
vpr flu.xml flu_core_content.vtr --blif_file flu_core_content.blif --timing_analysis on --timing_driven_clustering on --place_chan_width 74 --route_chan_width 74 --cluster_seed_type timing --seed 1 --place_algorithm path_timing_driven --nodisp --fix_pins vtr.pads

lvezier, 2016-12-15:
Here is my linux binary of VPR6
[vpr.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/654943/vpr.zip)

kmurray, 2016-12-15:
Thanks I will take a look.

I was using the VTR 7.0 release (VPR 7.0) and VTR 1.0 (VPR 6.0) available from [here](http://www.eecg.utoronto.ca/vtr/terms.html).

lvezier, 2016-12-15:
I have downloaded this release1.0, recompile it with graphics and I confirm that your VPR6 version behaves like VPR7.
I will check on my side.
Don't you think it is curious that these peripheral switch blocks are not symetrical ?

lvezier, 2016-12-15:
After investigation, I found a file fixed by my predecessor : 
[rr_graph2.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/655211/rr_graph2.zip)

kmurray, 2016-12-20:
The fix has now been committed to the development trunk (bc51e671d76a2044b816ec36b6f72652912a14a7).

VPR now produces the following for the periphery switch block, which is symmetric:

[image]

If you have any more issues or questions let us know.

closed 2016-12-20 by kmurray

lvezier, 2016-12-22:
Thanks for this fix.
For your information, there is no problem with waterfall, our port 8080 is blocked at my site.
On the other side, I am keeping to encounter difficulties to compile as static...

kmurray, 2017-01-02:
Regarding supporting newer versions of glibc on older releases of RHEL, you may find the following useful, as it lays out some of the different approaches supported by RedHat (in particular the section on the RedHat Developer Toolset):
https://developers.redhat.com/blog/2016/02/17/upgrading-the-gnu-c-library-within-red-hat-enterprise-linux/

lvezier, 2017-01-05:
Thanks for having made some research about this glibc issue, I will check that asap (I have to make VPR8 support operational first).