verilog-to-routing/vtr-verilog-to-routing #170: ODIN II Assertion Failure

kmurray, 2017-01-02:
ODIN II fails due to an assertion error (no error messages) when run on the attached circuit and architecture.

[test_case.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/681001/test_case.zip)

#### Expected Behaviour

ODIN should report a useful error message to aid in debugging the users verilog.

#### Current Behaviour

ODIN fails with an assertion (ODIN's internal `oassert`), which triggers a trap causing the program to core-dump when run normally, or pause if run in the debugger.

On my system (run in GDB) this produces:
```
Program received signal SIGTRAP, Trace/breakpoint trap.
get_name_of_pin_at_bit (var_node=0x85b090, bit=-1, 
    instance_name_prefix=0xae1d90 "top.memory_controller+memtroll")
    at /project/trees/vtr/ODIN_II/SRC/ast_util.c:602
602			return_string = make_full_ref_name(NULL, NULL, NULL, var_node->children[0]->types.identifier, rnode[2]->types.number.value+bit);
```

Where the failing assertion condition is actual on line 600 of ast_util.c:
```
		oassert((rnode[1]->types.number.value >= rnode[2]->types.number.value+bit) && bit >= 0);
```
where the condition `bit >= 0` fails.

#### Possible Solution

If the debugger is continued after the trap it produces the following error message:
```
ERROR: (File: ch_intrinsics_revised.v) (Line number: 315) This module port top.memory_controller+memtroll^memory_controller_address~0 is unused in module memory_controller
```
It may be that the assertion condition is to strong and can be loosened, but this will need to be verified by someone more familiar with ODIN.

#### Steps to Reproduce (for bugs)

1. Download the architecture and verilog test case attached
2. Use run_vtr_flow.pl to try to implement the circuit:
```
run_vtr_flow.pl k6_frac_N10_frac_chain_mem32K_40nm.xml ch_intrinsics_revised.v
```
ODIN II will fail
3. Move to the temporary working directory: `cd temp`
4. Run odin with gdb: 
```
gdb --args odin_II.exe -c odin_config.xml
```

#### Context

Reported by He Qi at University of Virgina, and reproduced by myself.

#### Your Environment

* VTR revision used: 880d066
* Operating System and version: Ubuntu 14.04 64-bit
* Compiler version: g++-5

jeanlego, 2019-06-22:
@ademmings. this should be relevant to the work you are doing right now.

sdamghan, 2019-09-09:
fixed as of right now

closed 2019-09-09 by jeanlego