verilog-to-routing/vtr-verilog-to-routing #172: IOB to track connection difference between VPR6 and VPR8

lvezier, 2017-01-05:
Following your advices, I am now using trunk dev version of VPR.

Our first issue was due to the naming of the NET: previously, block name and out name was the same. I fixed our bitstream generator.

Our second issue was due to trailing spaces in .net file, ex:
```
		<clocks>
			<port name="clk">top^fluclk_g </port>
		</clocks>
```
=> Fixed by a post treatment in perl to remove these trailing spaces.

Our last issue is that our bitstream generator is complaining about some impossible connection in SBOX. After investigation, it seems that track assignment has changed between VPR6 and VPR8.

#### Expected Behavior

Here is the GUI representation of IOB to SBOX nets with VPR6 (Block at (16,21) selected):
[image]
We see that first pad tracks are starting at the bottom.

#### Current Behavior

Here is the GUI representation of IOB to SBOX nets with VPR8 (Block at (16,21) selected):
[image]
We see that first pad tracks are starting at the middle.

#### Possible Solution

#### Steps to Reproduce (for bugs)

Here is the architecture file used for VPR8 (same as the previouus one provided in issue #167 :
[flu.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/687224/flu.zip)

Concerning the fc_in/out definition, I don't understand how to find the corresponding fractional value for absolute value = 16. I found empirically that 0.225 gives the same pattern.
Can you explain how to compute it?

#### Context

#### Your Environment

* VTR revision used: 8.0 dev
* Operating System and version: Linux RHEL7
* Compiler version:

kmurray, 2017-01-12:
**Routing Resouce Graph (RR-graph) differences with VPR 6**
We currently do not make any effort to ensure the routing resource graph (RR-graph)/switch patterns generated by VPR remain identical across revisions.  As we make changes/enhancements to the routing resource graph generator the pattern may change.

Since the VPR 6.0 release we have added a lot of flexibility to the RR-graph generator, so it is expected that the generated switch patterns will be different.

Your best approach is probably to dump out the RR-graph file for your architecture to a file and add support for loading that RR-graph back into VPR.  This is likely the only way to ensure the RR-graph stays the same across revisions.

**Fc Definition**
Fc is defined as the fraction of tracks in a channel that a pin connects to.

So at a channel width of 74, 16 connections corresponds to an Fc of 16/74 = 0.2162.  Using an Fc of 0.225 (0.225*74 = 16.65) is likely rounds to 16 connections by the RR-graph generator code.

Also note that absolute Fc specifications are supported.  See the <fc> tag [documentation](https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-%3Cfcdefault_in_type=).

lvezier, 2017-01-13:
Thanks for your answer, but it seems that there is not any easy way to make VPR 8.0 compatible with our FPGA architecture...
As I don't know very well VPR source code, could you show me where it makes sense to add this load/dump rr graph functionality ?

For your information, I successfully upgraded to VPR 7.0 (issue mentioned above appeared between VPR 7.0 and VPR 8.0) and generated bitstream seems to be OK. I just had to apply modification concerning switchblock (issue #167) and concerning pads assignment (issue #123).
Now, I call ABC before calling VPR, it improves our device utilization of around 10%.

To finish, I successfully compiled ABC and VPR for Windows using MingW gcc installed on Fedora (dnf install mingw32-gcc-c++.x86_64) but I had to modify some files:
 - add #define PCRE_STATIC in pcre/SRC/pcre.h under ifdef WIN32
 - substitutes "#ifdef _WIN32" by  "#if defined( _WIN32) && !defined(__MINGW32__)" in abc_with_bb_support/src/aig/rwt/rwtUtil.c and abc_with_bb_support/src/opt/rwr/rwrUtil.c
 - substitutes "restrict " variable name abc_with_bb_support/src/misc/espresso/unate.c by another name as "restrict" is a reserved keyword in last version of GCC
 - add "-static-libgcc -static-libstdc++" option to linker

kmurray, 2017-01-13:
Glad to hear you got things to compile statically.

The enhancements to the RR graph generator were added after the VTR 7.0 release, so it makes sense that the pattern differs in the latest code.

**Dumping the rr_graph**
If you run VPR with the [-echo on](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-echo_file) option it will generate a large number extra files dumping out VPR's internal state.  The main one of interest is `rr_graph.echo` which is a textual representation of the RR-graph that VPR generated.

**Loading the rr_graph**
The main function which creates the RR-graph is [build_rr_graph()](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/633a2fe5be0ed4843b3967a8dfe42fe90881db82/vpr/SRC/route/rr_graph.c#L221).  You will probably need to replace calls to build_rr_graph() with calls to your routine which initializes the RR-graph data structures from the previously generated file.

lvezier, 2017-01-13:
My issue with VPR 8 Linux version which needs RHEL 7 is still here. My add of  "-static-libgcc -static-libstdc++" option to linker was for Windows binary compilation. But we will probably support only RHEL 7 for future release embedding VPR 8.

Thanks for your info about load/dump, I will try that next week.

closed 2017-05-31 by kmurray