verilog-to-routing/vtr-verilog-to-routing #183: Cross-Compiling: error trying to make a static build

citylife4, 2017-04-25:
Hi! I'm trying to compile the latest VTR on a Ubuntu machine using gcc-4.9 to use it on another computer that doesn't have the latest flex and bison libraries. (flex version 2.5.4 and bison (GNU Bison) 2.4 )

But it fails when trying to link the builds with the following error:
[ 24%] Linking CXX executable ../../vpr/vpr
/usr/bin/ld: /usr/lib/gcc/i686-linux-gnu/4.9/libstdc++.a(eh_globals.o): undefined reference to symbol '___tls_get_addr@@GLIBC_2.3'
//lib/i386-linux-gnu/ld-linux.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

To try to make a static build i have add the flags to the compiler and linker: 
-static -static-libstdc++ -static-libgcc

also i have added the following commands to the CMakeList at the end of the file:
set(CMAKE_FIND_LIBRARY_SUFFIXES ".a")
#Set output locations to be in the main source tree under the relevant folder
#Static start
set_target_properties(<subdirectories> PROPERTIES LINK_SEARCH_START_STATIC 1)
set_target_properties(<subdirectories>PROPERTIES LINK_SEARCH_END_STATIC 1)

#### Your Environment

* VTR revision used: Latest
* Operating System and version: Linux lubuntu-vb 4.8.0-41-generic #44-Ubuntu SMP Fri Mar 3 15:26:20 UTC 2017 i686 i686 i686 GNU/Linux
* Compiler version: gcc-4.9 (Ubuntu 4.9.4-2ubuntu1) 4.9.4

Thanks!

kmurray, 2017-04-26:
I'm not sure what the linking issue is; I've never tried to build VTR against static system libraries. It looks like there is some incompatibility with your glibc versions.

One approach would be to generate the flex/bison parsers on a machine with appropriate flex/bison versions installed, and include them in the source files to be built.

citylife4, 2017-04-26:
Ok i'll try that!

Thanks!

closed 2017-04-26 by citylife4