verilog-to-routing/vtr-verilog-to-routing #184: Complex block with height of 2, connection to tracks changed from vpr7 to vpr8 (r6522)

jpalvadeau, 2017-05-10:
All CLB with height of 1 (Fc=16/74) seem to be correctly connected, whereas our flu_spu_24x24 (height = 2, Fc=14/74) connections to tracks differs from vpr7 to vpr8.

#### Expected Behaviour

Here is a screen-shot of connection shown in VPR7: 
[image]

#### Current Behaviour

Here is a screen-shot of connection shown in VPR8: 
[image]

#### Possible Solution

Should XML architecture description be modified/updated?
Is current repository version of VPR stable?

#### Context

Please find attached both XML used:
XML for VPR7 
[flu_vpr7_spread.xml.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/990155/flu_vpr7_spread.xml.zip)

XML for VPR8 
[flu_vpr8_spread.xml.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/990162/flu_vpr8_spread.xml.zip)

#### Your Environment

* VTR revision used: VPR8 (r6522)
* Operating System and version: RHEL 7
* Compiler version: gcc (GCC) 5.3.1 20160406

Thanks in advance.
Best regards,

Jeremy.

jpalvadeau, 2017-05-11:
Hi, I have notice a big improvement with the usage of custom pinlocation for SPU block.
XML file with custom pin location: [flu.xml.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/992303/flu.xml.zip)
To do this I disable some code in this file : libarchfpga/src/read_xml_arch_file.cpp
Check of custom pin location does not support a block with heigh of 2 (line 400), this seems to be an error. And also force user to specify location of clock (line 492), maybe a weakness of our architecture definition.

Now top, bottom and left side seems to be correctly (identically to VPR7) connected to tracks.
But on right side, connection to track pattern seems to be swapped between top right (offset=1) and bottom right (offset=0).

Comparison at offset 0 (vpr7 at left and vpr8 at right) 
[image]

Comparison at offset 1 (vpr7 at left and vpr8 at right)
[image]

VPR 8 track distribution at offset 1 is the same as VPR7 offset 0 and VPR8 track distribution at offset 0 is the same as VPR7 as offset 1.

kmurray, 2017-05-11:
Hi Jeremy,

Thanks for the detailed report!

**Switch Pattern Change**
This is an expected behavioural change with the latest VPR.  In typical commercial FPGA architectures it makes sense to allow blocks being 'overflown' by routing wires to connect to them (it helps spread out the routing demand of large blocks).

@sadegh68 made the change in c5dad5f948.  You could back out those changes if you want to get the old VPR7 style.

However that still will not guarantee that you get the same switch pattern.  We tweak the RR graph generation algorithms as needed to improve the generated architectures and/or add new features. As a result the exact switch patterns created change.

**Loading Fixed RR Graphs**
We are hoping to add support for loading and RR graph from an external file into VPR over the summer.

Once in place you should be able to 'save' a routing architecture to a file (or generate it with another tool) and then load it into VPR at a later time.  This would bypass VPR's RR graph generator and avoid the types of switch pattern changes you are seeing.

**Code Stability**
We generally try to keep the 'master' branch in a usable state, but it is under active development so there may be bugs/QoR issues.

We do run automated tests using our [BuildBot](http://builds.verilogtorouting.org:8080/waterfall?reload=60) after every code change.  

A way to check for any _obvious_ issues is to see if a revision has passed all the tests (basic, strong, nightly, weekly).  The 'basic' and 'strong' tests check basic functionality while the 'nightly' and 'weekly' test a broader range of benchmarks and architectures and are suitable for evaluating QoR. See [here](https://verilogtorouting.org/development/) for links to the benchtracker which allows you to view QoR metrics like critical path delay, wirelength and tool run-time/memory use.

However our test coverage is not complete, so even passing all tests does not guarantee that there are no bugs/QoR issues.

jpalvadeau, 2017-05-16:
Hi Kevin,
Thanks a lot for your helpful detailed answer!

With the change information you have provided: https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/c5dad5f948fac32bae7ae0a039624aaee232725a. 
We have only disable the "else{// for blocks with height > 1" in this file: vpr/SRC/route/rr_graph.c.
And with the usage of custom pinlocation for our SPU module, we have now the same  connection to tracks as VPR7!
We still have some assertions disabled in libarchfpga/src/read_xml_arch_file.cpp, we will try to improve our architecture definition before open a case about that.

I'm a little bit astonished about the different processing of pin distribution for a bloc with an height above 1.

Jeremy.

closed 2017-05-31 by kmurray