verilog-to-routing/vtr-verilog-to-routing #185: VPR8 STA reports details

jpalvadeau, 2017-05-16:
Hi,

New STA engine (tatum) of VPR8 provide a very user friendly repports. 
We are grateful for that!

But for developmental purpose the deeply detailed report of classic engine is useful. 
We try to enable #define ENABLE_CLASSIC_VPR_STA in vpr_types.h but we had compilation error.
After trying to fix those errors, we obtain a segmentation fault at end off execution of VPR.

Is it possible to have more detailed timing report with new engine?
Or could you check/correct the execution of classic engine (maybe deprecated?) ?

#### Your Environment

* VTR revision used: VPR8 (r6522)
* Operating System and version: RHEL 7
* Compiler version: gcc (GCC) 5.3.1 20160406

Thanks in advance.
Best regards,

Jeremy.

kmurray, 2017-05-16:
I've fixed the compilation and segfault issues with ENABLE_CLASSIC_VPR_STA in 2a44206.

However the classic STA engine should be considered deprecated; we plan to remove it in the (not to distant) future.

The new tatum-based STA engine operates on a timing graph derived from the user specified netlist (e.g. input BLIF file), so it only reports the aggregate delays between netlist primitives (LUTs, FFs, etc.).

What specific information are you missing from the new reports?

jpalvadeau, 2017-05-17:
Hi,
Thank for the fix!
It works well.
But sometime I obtain different timing between old and new sta engine. Is it normal? (new engine is stable or working?)

About the missing information or difference please refers to attached files: [flu_core_content.vtr.critical_path.out.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008328/flu_core_content.vtr.critical_path.out.txt) [report_timing.setup.rpt.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008329/report_timing.setup.rpt.txt)
For instance old report split in two timings the lut.out and the ble.out and the new report show only one line (with merged values).
Old report eases timing specification in architecture file.

I also notice a difference about path. 
In new report:
```
pin~0.inpad[0] (input)                                           0.000     2.000
nx42032z3.in[0] (names)                                          3.690     5.690
nx42032z3.out[0] (names)                                         0.202     5.892
```
but in the old one:
```
Node: 62  TN_INPAD_OPIN Block #3 (pin~0)
Pin: inpad.inpad[0] pb (pin~0)
[...]
Node: 61  TN_CB_OPIN Block #3 (pin~0)
Pin: io.inpad[0] pb (pin~0)
[...]
Node: 4  TN_CB_IPIN Block #0 (pout~0)
Pin: clb.I2[8] pb (pout~0)
[...]
Node: 27  TN_INTERMEDIATE_NODE Block #0 (pout~0)
Pin: ble.in[2] pb (nx42032z3)
[...]
Node: 32  TN_INTERMEDIATE_NODE Block #0 (pout~0)
Pin: lut4.in[2] pb (nx42032z3)
[...]
Node: 37  TN_PRIMITIVE_IPIN Block #0 (pout~0)
Pin: lut.in[2] pb (nx42032z3)
[...].
```

The new report does not indicate the same clb/ble/lut inputs. 
Is this normal?

Jeremy.

kmurray, 2017-05-17:
# Delay Differences
The new and old STA engines should report the same timing information.  However they present it differently.

The old STA engine reported delays between each *physical* element of the FPGA (external routing, intermediate levels of PB heirarchy etc.).

The new STA engine always reports delays between *logical* elements (i.e. between elements of the user-specified primitive BLIF netlist).

In your example the connection delays between ``pin~0.inpad`` and ``nx42032z3.in`` are equivalent.

The new STA engine reports the aggregate delay of **logical connections** in the primitive netlist (where ``pin~0`` and ``nx42032z3`` are names in the BLIF netlist):
```
pin~0.inpad[0] (input)                                           0.000     2.000
nx42032z3.in[0] (names)                                          3.690     5.690
nx42032z3.out[0] (names)                                         0.202     5.892
```

While the old STA engine reports delays for each **physical connection** in the FPGA (e.g. ``ble`` is an internal hierarchy of the logic block which does not appear in the BLIF):
```
Node: 63  TN_INPAD_SOURCE Block #3 (pin~0)
Pin: inpad.inpad[0] pb (pin~0)
T_arr: 0  T_req: 9.2916e-08  Tdel: 2e-09
Internal Net: pin~0.  Pins on net: 2.

Node: 62  TN_INPAD_OPIN Block #3 (pin~0)
Pin: inpad.inpad[0] pb (pin~0)
T_arr: 2e-09  T_req: 9.4916e-08  Tdel: 5e-10
Internal Net: pin~0.  Pins on net: 2.

Node: 61  TN_CB_OPIN Block #3 (pin~0)
Pin: io.inpad[0] pb (pin~0)
T_arr: 2.5e-09  T_req: 9.5416e-08  Tdel: 1.67e-09
External-to-Block Net: #4 (pin~0).  Pins on net: 2.

Node: 4  TN_CB_IPIN Block #0 (pout~0)
Pin: clb.I2[8] pb (pout~0)
T_arr: 4.17e-09  T_req: 9.7086e-08  Tdel: 1.52e-09
Internal Net: pin~0.  Pins on net: 2.

Node: 27  TN_INTERMEDIATE_NODE Block #0 (pout~0)
Pin: ble.in[2] pb (nx42032z3)
T_arr: 5.69e-09  T_req: 9.8606e-08  Tdel: 0
Internal Net: pin~0.  Pins on net: 2.

Node: 32  TN_INTERMEDIATE_NODE Block #0 (pout~0)
Pin: lut4.in[2] pb (nx42032z3)
T_arr: 5.69e-09  T_req: 9.8606e-08  Tdel: 0
Internal Net: pin~0.  Pins on net: 2.

Node: 37  TN_PRIMITIVE_IPIN Block #0 (pout~0)
Pin: lut.in[2] pb (nx42032z3)
T_arr: 5.69e-09  T_req: 9.8606e-08  Tdel: 2.02e-10
Internal Net: pin~0.  Pins on net: 2.
```

The new STA engine reports a delay of: 3.690ns
The old STA engine reports a delay of (sum of Tdel for Nodes 62, 61, 4, 27, 32):  5e-10 + 1.67e-9 + 1.52e-9 + 0 + 0 = 3.690e-9 = 3.690ns

These are equivalent.

Additionally, ENABLE_CLASSIC_VPR_STA currently checks that the old and new STA engines always report the same critical path delay.

# Pin differences
The pin differences (e.g. ``nx42032z3.in[0]`` in the new STA engine versus ``lut.in[2] pb (nx42032z3)`` in the old), is expected.

VPR's packer may re-order LUT inputs (which are logically equivalent provided the LUT mask is updated) as it routes the internal connections within a PB.

This re-ordering is reflected in the pin numbers reported by the old STA engine. 

The new STA Engine reports the pins as they were logically connected in the primitive BLIF netlist (i.e. without re-ordering).

jpalvadeau, 2017-05-18:
Hi,
Thank you for all those informations.
I will continue tu use both engines during timing specification phase.
If I found again a different path and timing between both engines, i will notice you.

Jeremy.

closed 2017-05-18 by kmurray