verilog-to-routing/vtr-verilog-to-routing #186: Fixed Architecture in different runs of VTR

citylife4, 2017-05-17:
I'm trying to test there is a way to use the same architecture in two different runs of the program.
To do this i'm using two verilog files with the same inputs and outputs and first i'm running the larger one, and the vpr step runs with the following command:

vpr/vpr arch_complex_switch_new.xml fsm8_16_13 --blif_file fsm8_16_13.pre-vpr.blif --timing_analysis on --timing_driven_clustering on --cluster_seed_type blend --seed 1 -echo_file on -sweep_dangling_primary_ios off -sweep_dangling_nets off -disp on 

[fsm8_16_13_last.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008156/fsm8_16_13_last.v.txt)

[arch_complex_switch_new.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008169/arch_complex_switch_new.xml.txt)

then the small one runs with a a architecture file that assembles that one but with the size, pads, and width fixed.

Command:
vpr arch_complex_switch_new_fixed.xml fsm8_16_13 --blif_file fsm8_16_13.pre-vpr.blif --timing_analysis on --timing_driven_clustering on --cluster_seed_type blend --seed 1 -sweep_dangling_nets off -echo_file on -sweep_dangling_primary_ios off -route_chan_width 16 -fix_pins fsm8_16_13.pads -disp on

[fsm8_8_16_first.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008158/fsm8_8_16_first.v.txt)
[arch_complex_switch_new_fixed.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008177/arch_complex_switch_new_fixed.xml.txt)
[fsm8_16_13.pads.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1008180/fsm8_16_13.pads.txt)

Thank you

kmurray, 2017-05-17:
I don't quite follow what you are trying to do.

Are you trying to size your FPGA based on the fsm_8_16_13 design, and then implement fsm_8_8_16 onto the same device, using the pad locations from the fsm_8_16_13 run?

citylife4, 2017-05-17:
Yes, thats my idea! And its working but there is some differences in the wires (the two runs produce different rr_graph.echo files)

kmurray, 2017-05-17:
OK, I would probably try the following:

1. Run VPR to auto-size the FPGA to the first design. Note down the device grid size and your desired channel width.
2. Re-run VPR targeting the fixed device grid size and channel width on the first design (this should ensure you are targeting the same fixed device).
3. Run the second design using the fixed device grid, channel width and pads locations from (2).

citylife4, 2017-05-17:
In my approach i'm only doing the 1st and 3rd points.
Then if i'm understanding correctly what I need to do differently,  is to run the first design again with a fixed device grid size (using the above fixe.xml file) and then run with the same commands/arguments and files but using the new design.

closed 2017-05-29 by citylife4