verilog-to-routing/vtr-verilog-to-routing #187: Mixed sequential/combinational block timing modelling

jpalvadeau, 2017-05-18:
Hi,

I try to define following bloc:
[image]

I think following description is correct:
(it is an extract of all our spu modes description) 
```
<model name="spu_acbrrr">
      <input_ports>
        <port name="a"        combinational_sink_ports="out"/>
        <port name="b"        clock="clk" combinational_sink_ports="out"/>
        <port name="clk"    is_clock="1"/>
      </input_ports>
      <output_ports>
        <port name="out"        clock="clk"/>
      </output_ports>
</model>

<pb_type name="spu_acbrrr" blif_model=".subckt flu_spu_acbrrr" num_pb="1">
    <input  name="a"       num_pins="24"/>
    <input  name="b"       num_pins="24"/>
    <clock  name="clk"     num_pins="1" port_class="clock"/>
    <output name="out"     num_pins="48"/>
            
    <delay_constant max="6.2e-9" in_port="spu_acbrrr.a"       out_port="spu_acbrrr.out"/>
    <delay_constant max="6.2e-9" in_port="spu_acbrrr.b"       out_port="spu_acbrrr.out"/>
    <T_setup      value="2.5e-9"    port="spu_acbrrr.out"     clock="clk"/>
    <T_clock_to_Q   max="0.5e-9"    port="spu_acbrrr.out"     clock="clk"/>
    <T_setup      value="2.5e-9"    port="spu_acbrrr.b"     clock="clk"/>
    <T_clock_to_Q   max="0.5e-9"    port="spu_acbrrr.b"     clock="clk"/>
</pb_type>
```

But with this description I have following error from vpr:
`Error 1: STA Engine: IPIN nodes should only drive OPIN nodes`

This error disappears if y specify:
`<port name="a"        clock="clk" combinational_sink_ports="out"/>`
But I do not want a register on input a.

If y disable check and assertion in TimingGraph.cpp line 704, VPR seems to works fine and timing analysis seems to be correct.

Should VPR source code be improved? Or my description is incorrect or unsupported?

#### Our Environment

* VTR revision used: VPR8 (09a2d24)
* Operating System and version: RHEL 7
* Compiler version: gcc (GCC) 5.3.1 20160406

Thanks in advance.

Jeremy.

kmurray, 2017-05-18:
This was a VPR issue. It should be fixed in 8c6fd39.

The assertion was overly aggressive, and I had never tested this case. I've relaxed it to allow either OPIN or SOURCES as the sinks of an IPIN.

I've also updated the timing [modelling tutorial](https://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#sequential-block-with-internal-paths-and-combinational-inputl) with an example of this type of block.

closed 2017-05-18 by kmurray

jpalvadeau, 2017-05-18:
It works.
Thanks a lot.

Jeremy.