verilog-to-routing/vtr-verilog-to-routing #194: canot get the power estimation result when using blif file

mafuang, 2017-06-01:
I tried to get the power estimation result on blif file when using architecture from timing destination but it did not work. Thus, I added the following command on the run_vtr_flow.pl

./run_vtr_flow.pl ../benchmarks/blif/alu4.blif ../arch/timing/k4_N4_90nm.xml -power -cmos_tech ../tech/PTM_22nm/22nm.xml -starting_stage vpr

However, the result could not be run as it appeared 

k4_N4_90nm/alu4. . .failed: vpr

The verilog file works fine with the power estimation without -starting_stage vpr, but the blif file it has to start at the vpr stage, so I added -starting_stage vpr; however, it doesn't work. I tried to find the way to solve this problem from both the manual and the internet for a while, but I haven't found anything I could do. 
Do I have to add or change anything to get the result? 

My OS is Ubuntu 15.10
My VTR is 7.0.7

jgoeders, 2017-06-01:
Thanks for the report.  I'll look into this and get back to you today or tomorrow.

kmurray, 2017-06-01:
One thing to note is that providing `-starting_stage vpr` skipps ACE, the tool which does activity estimation. Perhaps that is part of the problem?

jgoeders, 2017-06-01:
There are actually a number of errors that you are running into:

If you look at temp/vpr.crit_path.out, you can see the error that is produced:
> Error 1: k4_N4_90nm.xml:7 Missing required child node 'power' in parent node 'architecture'
Sorry the error is not more descriptive, but the problem is that the architecture file you selected does not contain power-related information.  

If you instead select an architecture file with power information (such as k6_N10_40nm.xml), you will get the following error:
> Error 1: /home/jgoeders/vtr/libvtrutil/src/vtr_util.cpp:287 Error opening file alu4.act for r access: No such file or directory.

In this case it is reporting the problem that Kevin suggested, that you must run activity estimation before power analysis.  The error indicates that the activity file (.act) is missing. You can take a look at the power documentation https://docs.verilogtorouting.org/en/latest/vtr/power_estimation/ for more information about the activity estimation.  

If you change the starting_stage to ace (the activity estimation tool), you should see:
> k6_N10_40nm/alu4...failed: ace

And if you look at temp/ace.out you will see the following error:
> No clocks detected in blif file.  This is not supported.
 
The activity estimation tool requires that the circuit be synchronous.  You can see #95 for a full description as to why this requirement is there, and a possible workaround (adding a clock signal, and adding registers to inputs and outputs)

mafuang, 2017-06-18:
Thank you very much for your suggestions. However, I'm still confused with the way to force the fake clock rate for power estimation. I'm not sure if that means I have to add the fake clock rate into the .blif file or there are some other ways to do it (like doing something with this command "$VTR_ROOT/ace2/ace -b <abc.blif> -o <activities.act> -n <new.blif>" ? )
Could you please let me know how or where can I find the way to add the fake clock rate into the file?

jgoeders, 2017-06-19:
I don't know of any way to automatically add a clock and input/output registers to your design.

You will have to manually modifiy the BLIF, or write some script that would:
1) Add a clock to the design
2) Add a flip flop to the design for every input and output.  The FFS at the input need to be driven by the inputs, and the FFs at the output need to drive the outputs.  Connect the clock to these flip flops.
3) For every LUT where an input is used, you will need to replace this to use the output of the appropriate FF.
4) For every LUT that drives an output, you will need to instead drive the appropriate FF.

closed 2017-07-25 by jgoeders