verilog-to-routing/vtr-verilog-to-routing #209: build bot fail on qor nightly.

jeanlego, 2017-06-28:
regression_tests/vtr_reg_nightly/vtr_reg_qor_chain...[Fail] k6_frac_N10_frac_chain_mem32K_40nm.xml/LU8PEEng.v 
num_pre_packed_nets: 
result = 40355 
golden = 36217

larger number of prepacked nets is bad or good? someone with better knowledge of the full flow can 
help with that?

kmurray, 2017-06-28:
This is basically saying that the number of nets produced by ODIN has increased from 36217 to 40355.

It would be good to know what in ODIN to caused this change.

kmurray, 2017-06-28:
It looks like this is effecting both the nightly and weekly QoR regression tests.

Using git bisect it looks like the offending commit is:
```
b09fd0d90dc2f92fe855d2e0f30c82644e9f697a is the first bad commit
commit b09fd0d90dc2f92fe855d2e0f30c82644e9f697a
Author: jeanlego <jeanphilippe.legaul@gmail.com>
Date:   Mon Jun 5 02:24:13 2017 +0000

    added a strcmp inside @vtr_util to do NULL check first(undefined in c)
```

In revision b09fd0d and later Odin produces a different different BLIF output file for the k6_frac_N10_frac_chain_mem32K_40nm.xml/LU8PEEng.v circuit architecture combination, which has more nets than before.

I've reverted the change on the master branch to fix the regression tests (6aff65e). Once you've fixed the bug we can consider re-applying it.

jeanlego, 2017-06-30:
strcmp overloading was more a temporary fix until i move everything to std::string. I will mark this as closed. also, to prevent future breakage of master I was wondering if it was possible to have a stable branch and a dev branch? if a dev build passes it gets merged onto stable. considering these test take a while to run on their own?

closed 2017-06-30 by jeanlego