verilog-to-routing/vtr-verilog-to-routing #212: Timing report breakdown routing delay and logic delay

heiner-bauer, 2017-07-12:
With the new STA, critical path delay breakdown into routing and logic delay has been removed.

#### Expected Behaviour
Report the routing delay and logic delay in addition to the total delay of the critical path, and possibly output the timing along the critical path in terms of physical blocks.

#### Current Behaviour
Timing reports in terms of the BLIF netlist provide almost no insight because intermediate nodes are hard to correlate with the design (due to ABC's node naming scheme).
The only way to get back the critical path breakdown is to enable the classical STA, which I assume, will be removed in the future.

#### Context
 Delay breakdown is important to determine whether a certain benchmark or architecture is dominated by routing or logic delay. This is crucial when trying to improve either the routing or logic block architecture. The classical report of the critical path (<design>.critical_path.out) was also helpful in verifying delays of physical blocks to ensure correct timing annotation in the architecture description.
This is also hinted at in issue #185 .

#### Your Environment
* VTR revision used: 8.0.0-dev
* Operating System and version: GNU Linux-2.6.32 x86_64
* Compiler version: GCC 4.9.4

kmurray, 2017-07-17:
I agree having a more detailed report would be useful. We will hopefully add a more detailed report which breaks down in intra-block delays (the classic STA report already lumped together the inter-block wire delays) in the future.

However, I would point out that the logic (e.g. LUT, FF) and routing delays can already be derived from the existing timing reports; the key difference from the classic STA engine is that the intra-block and inter-block routing delays are lumped together.

heiner-bauer, 2018-05-18:
Thanks, that is definitely great, however, would it also be possible to report the sum of logic vs. routing delay for each path ? This measure would be more useful for general architectural optimizations. I assume the STA already has an internal representation to differentiate the two, so parsing and reinterpreting (by an external script) the timing report could be avoided.