verilog-to-routing/vtr-verilog-to-routing #215: Clean-up Odin Related GCC Warnings

kmurray, 2017-07-17:
#### Expected Behaviour
VTR should compile cleanly without warnings using GCC.

#### Current Behaviour
Recent changes to Odin introduced a large number of warnings when compiled in debug mode.

A small selection:
```
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_elaborate.cpp:1430:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_elaborate.cpp:1503:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_elaborate.cpp:1607:52: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:235:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:285:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:331:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:460:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1288:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1447:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1543:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1545:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1605:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1607:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:1717:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/parse_making_ast.cpp:2141:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:62:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:85:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:166:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:296:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:348:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:356:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:819:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:884:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_ast/ast_util.cpp:944:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_blocks/adders.cpp:1150:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/odin_util.cpp:573:13: warning: no previous declaration for ‘std::string base_log2_helper(char, size_t)’ [-Wmissing-declarations]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:175:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:180:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:213:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:225:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:243:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:248:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:329:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:426:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:453:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:495:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/home/vtrbot/vtr-buildbot/slave_basic/basic/build/ODIN_II/SRC/lib_netlist/netlist_create_from_ast.cpp:499:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
```

See [here](http://builds.verilogtorouting.org:8080/builders/basic/builds/167/steps/make/logs/warnings%20%2891%29) for the full list from BuildBot.

#### Possible Solution
Most of these issues are related to using mismatched signed/unsigned types. The appropriate type should be used, and other warning related issues resolved.

#### Steps to Reproduce (for bugs)
Run the following from the VTR root directory:
```shell
$ make clean
$ make odin_II
```

#### Your Environment
* VTR revision used: dcd0d79
* Operating System and version: Ubuntu 14.04
* Compiler version: gcc 4.9.4

kmurray, 2017-07-25:
Fixed as of 8c41c32

closed 2017-07-25 by kmurray

mozammalhossain, 2017-07-25:
I will have a look.

Thanks.
Mozammal

aarongraham9, 2017-07-25:
No need; fixed as of 8c41c32 and verified in Nightly Regression Test Run [`118`](http://builds.verilogtorouting.org:8080/builders/nightly/builds/118).