verilog-to-routing/vtr-verilog-to-routing #216: Fix Odin regression failure relating to large constants

kmurray, 2017-07-17:
#### Expected Behaviour
Odin should be able to handle arbitrary length constants.

#### Current Behaviour
Odin fails for long constants as a result of #214.
```
ERROR: (Line number: 2178) This base 2 number (00101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010) is too long for Odin
```

#### Steps to Reproduce (for bugs)
From the VTR root:
```shell
$ ./ODIN_II/odin_II -a vtr_flow/arch/timing/EArch.xml -V vtr_flow/benchmarks/verilog/mkDelayWorker32B.v
```

and/or

```shell
$ ./ODIN_II/odin_II -a vtr_flow/arch/timing/EArch.xml -V vtr_flow/benchmarks/verilog/LU32PEEng.v
```

#### Context
This issue is causing a failure of the [nightly regression tests](http://builds.verilogtorouting.org:8080/builders/nightly/builds/113). As a result of merging #214.

The relevant changes have been reverted in b319fb8.  Once the code is fixed, it can be re-merged (see: 'Reverting the Revert' [here](https://git-scm.com/blog/2010/03/02/undoing-merges.html)).

#### Your Environment

* VTR revision used: dcd0d79
* Operating System and version: Ubunutu 14.04
* Compiler version: gcc 4.9

jeanlego, 2017-07-25:
hello! the reason why i make it fail is that , while technically it used to pass without saying anything, odin would convert binary values of undounded length into long long, while it would put out some result. they would not be accurate. the only way to fix this is to use

kmurray, 2017-07-25:
I agree we ultimately want Odin to handle large constants correctly (e.g. via a large number library like GMP). Particularly since some of our regression circuits (LU32PEEng, mkDelayWorker32B) already make use of such constants.

However, until such support is stabilized I'd prefer to keep the regression tests clean, so we can detect any new errors in downstream tools (e.g. VPR).

I think it is probably best to keep this issue open to track the fundamental issue.

jeanlego, 2017-07-25:
i looked into GMP and boost (wich would help for a whole lot of things too) but the licencing would break ours?? (i might be totally wrong on this someone knows??) hence why i made changes to only play with binaries and started writing a library for it (probalby very slow since it would handle them based on the basic verilog defined truth table) but also would allow ODIN to parse x and z binaries, they are not usable down the line but since they are defined I just wanted ODIN the be able to parse them.

Would it be practical to have ODIN   have flag that throws only a warning to the lines having x or z binaries and let the user decide to which value to translate them to? or maybe like a --strict flag that would prevent ODIN from accepting x and z otherwise set them as 0 but popup a clear warning ?? any use for that or just 0-1 and use boost or GMP (if licencing is okay??)

jeanlego, 2018-06-16:
small update. I've added a proper warning for this in Odin but let the compiler run nonetheless. there's TODO scattered all over for this issue for anyone who wants to step in and help out.

jeanlego, 2019-07-11:
fixed

closed 2019-07-11 by jeanlego