verilog-to-routing/vtr-verilog-to-routing #220: The <gridlocations> and <loc> tags can't describe the layout of an Artix-7 or iCE40

jhol, 2017-07-23:
http://docs.verilogtorouting.org/en/latest/arch/reference/#tag-%3Cgridlocations%3E

The `<loc>` tag provides `type=` options of `col`, `rel`, `fill` and the undocumented `perimeter` option.

This isn't enough to describe the layout of an Xilinx Artix-7 or an Lattice iCE40

Clifford Wolf has been reverse engineering the FPGA layout in his "Project X-Ray", and has produced this [tile-browser](http://scratch.clifford.at/prjxray/grid-xc7a50tfgg484-1-db.html) for the XC7A50. (NOTE: Currently only works in a Google Chrome and Chromium).

It also isn't quite enough to describe the layout of the Lattice iCE40 FPGAs. Clifford Wolf has reverse engineered the layout in his [IceStorm project](http://www.clifford.at/icestorm). The iCE40 has a few [special tiles](http://www.clifford.at/icestorm/io_tile.html) (PLL cores, "WARMBOOTs") which are located adjacent to the the main FPGA fabric.

#### Expected Behaviour
`<loc>` should have enough `type=` options to describe real-world FPGAs.

#### Current Behaviour
The current selection of types is quite limited. 

`rel` seems like an odd design for a `type` option. Is it useful for research purposes? Doesn't seem very useful for real-world FPGA layouts.

#### Possible Solution
More `<loc>` types are needed such as `rect`, `row`, `single`, or anything equivalent to these that can fully describe the tile locations in real-world FPGAs.

#### Context
 The only open source PAR for iCE40 is [arachne-pnr](https://github.com/cseed/arachne-pnr) which is hard coded for iCE40s only. The Artix-7 bitstream reverse engineering is still a work in progress, and no Open Source PAR exists yet. The Yosys/Open Source FPGA community is interested in using VPR as a PAR for open source FPGA synthesis for Artix-7 and iCE40.

kmurray, 2017-07-24:
This would make an excellent enhancement to VTR, as this has been a known limitation for a long time. (We just haven't had the motivation to add support for it before).

I do not believe there is any fundamental limitation preventing this. It would involve:
1. Updating the architecture XML format and libarchfpga to support more general location specifications
2. Updating the grid construction code in VPR to interpret and apply the extended specifications when building the device grid

Some of this work has already been done by [VTR-to-bitstream](https://eddiehung.github.io/vtb.html), but was never merged back into the VTR mainline.

kmurray, 2017-07-24:
To follow up on some of the other points:

> rel seems like an odd design for a type option.

Yes, this is motivated by FPGA architecture research, where we do not (necessarily) want to limit ourselves to a fixed/specific device layout.

> The Yosys/Open Source FPGA community is interested in using VPR as a PAR for open source FPGA

This makes a lot of sense to me, since VPR is already architecture agnostic.  Once all the features required to describe the target FPGA device were added the pack/place/route algorithms should just work.

kmurray, 2017-08-03:
I've updated libarchfpga and VPR to support the construction of arbitrary device grids.

**New Location Specifications**

The device layout can now be specified under the `<layout>` tag with either:
- A single `<auto_layout>` tag for an automatically sized device, or
- One or more `<fixed_layout>` tags which define fixed device layouts (e.g. a device family).

By default VPR will pick the smallest device which fits the circuit being implemented (although this can be controlled with the `--device` option).

We also now support a variety of different location specifications including:
- `<fill>`
- `<perimeter>`
- `<corners>`
- `<single>`
- `<col>`
- `<row>`
- `<region>`

which support both absolute and relative location specifications.
The [updated documentation](https://docs.verilogtorouting.org/en/latest/arch/reference/#grid-location-tags) includes examples. 

Most of the location specifications tags are just syntactic sugar, and libarchfpga converts them to `<region>`s before passing them on to VPR. 

**Current Limitations**
* This has been tested with the classic VPR architecture layouts which include perimeter I/O and column based layouts. Non-column based layouts have seen less testing (due to a lack of test cases).

* VPR's routing resource graph generator currently does not generate routing channels at the edges of the device (a legacy of always assuming perimeter I/O). As a result non-I/O blocks may not be routable if placed around the perimeter. This should not be a problem if an [explicit RR Graph file](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-file-format-xml) is loaded, but will need to be fixed for VPR-generated RR Graphs.

kmurray, 2017-09-06:
I've updated the RR Graph generator to support blocks with width > 1, and tested more extensively with non-perimeter I/O and non-column based architectures (fixing issues where appropriate).

I believe that should be enough to model the device floorplans of interest. 

If you encounter any other issues please file a bug!

closed 2017-09-06 by kmurray