verilog-to-routing/vtr-verilog-to-routing #227: stratixiv_arch.timing.xml cannot be found for vtr_reg_strong

jeanlego, 2017-08-04:
regression_tests/vtr_reg_strong/strong_titan
-----------------------------------------
Architecture file not found (/workspace/vtr-verilog-to-routing/vtr_flow/arch/titan/stratixiv_arch.timing.xml) at scripts/run_vtr_task.pl line 308, <CONFIG_FH> line 182.

Parsing test results... 
scripts/parse_vtr_task.pl -l /workspace/vtr-verilog-to-routing/vtr_flow/tasks/regression_tests/vtr_reg_strong/task_list.txt  
Unknown experiment id at scripts/parse_vtr_task.pl line 535.

kmurray, 2017-08-04:
Have you run `make get_titan_benchmarks` from the VTR root directory? That will download the appropriate benchmarks/architectures and extract them into the VTR source tree.

We don't include the titan benchmarks since they are too large (~10GB) to put in revision control.

jeanlego, 2017-08-04:
must be new. used to simply run vtr_reg_strong without a hitch from the container . thanks! sorry for the disturbance.

kmurray, 2017-08-04:
No worries. Previously we didn't test the Titan benchmarks in vtr_reg_strong. 

I added the titan sub-test since there have been a couple titan related bus which slipped through the strong regression and were only caught by the nightly/weekly tests.

closed 2017-08-04 by kmurray