verilog-to-routing/vtr-verilog-to-routing #235: Can I customize fpga architecture mentioned in xml file. And does connection patterns in the switch box are decided by vpr?

sgupta020, 2017-09-13:
#### Expected Behaviour
I am running vpr for fpga architecture file "k6_N10_mem32K_40nm.xml" and giving channel width as 40.
(1)  I want two continuous tracks in INC direction, then next one track in DEC direction. And so on. In other words I want following:
   Track numbers 0 and 1 on INC direction. Track number 2 on DEC direction.
   Track numbers 3 and 4 on INC direction. Track number 5 on DEC direction.
   Track numbers 6 and 7 on INC direction. Track number 8 on DEC direction.
   So on.

   Is it possible to achieve this? If answer is yes, then How we can achieve this? Is it possible to achieve only by modifying xml file?
(2) I want to decide connection patterns inside the switch box. (Suppose I decide some patterns inside switch box. Will it be possible to support that particular pattern inside switch box through architecture xml file?)

   Is it possible to achieve this? If answer is yes, then How we can achieve this? Is it possible to achieve only by modifying xml file?
(3) Does all the switch boxes in a fpga will have same patterns inside?

So in short for above three questions, Is it possible to give any routing architecture via modifying xml file?

#### Current Behaviour
I am running vpr for fpga architecture file "k6_N10_mem32K_40nm.xml" and giving channel width as 40.
(1)  One track in INC direction, then one track in DEC direction. In other words:
   Track number 0 on INC direction. Track number 1 on DEC direction.
   Track number 2 on INC direction. Track number 3 on DEC direction.
   Track number 4 on INC direction. Track number 5 on DEC direction.
   So on.

(2) Architecture xml file and/or vpr tool deciding the connection patterns in the switch box.

(3) All the switch boxes in a fpga are NOT having same patterns inside.

kmurray, 2017-09-13:
**(1)** This is currently not supported by VPR since the RR graph generator assigns tracks in pairs of INC and DEC.

It may be possible to make VPR support your type of routing architecture by either:
a) Modifying VPR's RR graph generator (rr_graph.cpp and related) to support generating the type of architecture you desire, or
b) Generate an [external RR graph file](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-file-format-xml) (e.g. with an external tool) which describes your routing architecture

Either of the above approaches would involve a _significant amount of work and debugging_ and I would not recommend them unless you have extremely strong motivations for modelling this type of architecture.

**(2)** VPR supports specifying [custom switch block patterns](https://docs.verilogtorouting.org/en/latest/arch/reference/#custom-switch-blocks) in the architecture file.

**(3)** With custom switch blocks you gain some control of where different switch block patterns are applied on the device.

sgupta020, 2017-09-22:
Good explanation.

closed 2017-09-22 by sgupta020

Wkraipak, 2022-10-22:
Hi Sgupta,
I am also working on adding the custom switch block patterns in the architecture file. wanted to connect with you and understand how you have done it.
Wasim

Wkraipak, 2022-10-22:
Kmurray,
do you have any example of adding a custom switch box in the Fpga architecture xml file. what if we have dedicated connection patterns inside the switch box how to describe them in the xml file

Wasim