verilog-to-routing/vtr-verilog-to-routing #237: Difference in ODIN generated blif files running with and without architecture xml file

sgupta020, 2017-09-22:
#### Expected Behaviour

#### Current Behaviour
(a)  Run ODIN II by giving fpga architecture file as argument.
(b)  Run ODIN II without giving fpga architecture file.
What is the difference between generated blif files in case (a) and case (b)?

#### Possible Solution

#### Steps to Reproduce (for bugs)

1. 
2. 
3. 
4. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

kmurray, 2017-09-22:
ODIN is not really my area.

However I suspect that (a) will synthesize to both logic (LUTs) and the hard primitives supported by the architecture (adders, multipliers, RAMs etc.), while (b) will synthesize to just logic (LUTs).

@KennethKent @PanosNB: Perhaps one of the ODIN developers can comment in more detail?

sgupta020, 2017-09-22:
Thanks for reply.
(1) Who will decide size of the LUTs used? [in the case (a)]
(2) Who will decide size of the LUTs used? [in the case (b)]
(3) In case of running ODIN with architecture file I found that primitive dual_port_ram is of 7 inputs. But in the generated blif file used primitive dual_port_ram is of different number of inputs. So, from where primitive dual_port_ram is picked up in the blif file? Is this primitive taken from architecture xml file or it is some generic primitive?

Thanks

KennethKent, 2017-09-22:
Kevin is correct. To answer your other questions.

1 & 2) LUT size is not considered by Odin. It generates a netlist and downstream tools will deal with packing into LUTs
3) Odin learns about dual_port_ram definitions from the architecture xml file. If it is not defined there then the assumption is that the FPGA does not have hard memories.

sgupta020, 2017-09-25:
Hi,
So, till now following is my understanding (I have some doubts inlined):-
**(1) ODIN:-** It does synthesis.
We can run ODIN II by following two ways: 
(A) Run ODIN by giving fpga architecture file as argument:-
While synthesizing it uses following:
(i) hard blocks (adders, multipliers, memories etc.) defined in the xml file
(ii) generic logic components (These generic logic components are only LUTs or these can be other components also? If other components are also possible then please give some examples of other components. Are “.names” and “.latch” generic logic components?)

(B) Run ODIN WITHOUT giving fpga architecture file:-
While synthesizing it uses only generic logic components. (These generic logic components are only LUTs or these can be other components also? If other components are also possible then please give some examples of other components. Are “.names” and “.latch” generic logic components?)

**(2) ABC:-**  ABC does synthesis and mapping both.
In my case running VTR script uses following command to run abc:-
vtr_release/vtr_flow/../abc_with_bb_support/abc -c read diffeq1.odin.blif; time; resyn; resyn2; **if -K 6**; time; scleanup; time; scleanup; time; scleanup; time; scleanup; time; scleanup; time; scleanup; time; scleanup; time; scleanup; time; scleanup; time; write_hie diffeq1.odin.blif diffeq1.abc.blif; print_stats

Question (1):  Does ABC do resynthesis? Why it does because ODIN has already done synthesis. If ABC does synthesis, then How it is different than synthesis done by ODIN?

Question (2): In the above command argument passed is “if -K 6”. Does it mean that ABC does mapping of generic logic components to 6-input LUTs? Or it can map to 5-input LUTs and 6-input LUTs?

Question (3):  Suppose ODIN was run by giving fpga architecture file as argument.
Now while running ABC:
(i) hard blocks (adders, multipliers, memories etc.) present in the odin.blif will be not touched?
(ii) generic logic components  present in the odin.blif will be mapped to 6-input LUTs?

Question (4): Suppose ODIN was run WITHOUT giving fpga architecture file.
Now while running ABC:
 generic logic components  present in the odin.blif will be mapped to 6-input LUTs?

Question (5): ABC does technology dependant or technology in-dependent mapping?

Question (6): Does ABC uses “cadence.genlib” library? I have not found this file in the VTR tool downloaded.

KennethKent, 2017-09-25:
Regarding ODIN, I suggest that you read about the Berkeley Logic Interchange Format (BLIF). Effectively, ODIN generates a netlist that is in this format. When you read the online pdf describing the format you will see that .names and .latch are standard. Regardless of how you run ODIN, you get this. If you specify an xml architecture file then you get additional components via "subckt". ODIN retrieves the models from the FPGA architecture file and can then use subckt for these models in the blif output.

kmurray, 2017-09-25:
To follow up on some of the other points:

ABC performs both logic optimization (technology independent), and technology mapping (technology dependent). By default the VTR scripts instruct ABC to map to LUTs.  

For more details on the exact ABC commands I would suggest looking at ABC's documentation.

closed 2017-09-26 by kmurray