verilog-to-routing/vtr-verilog-to-routing #242: Allow architecture to specify the connection boxes from specific segments to specific tile ports

mithro, 2017-11-09:
# Allow architecture to specify the connection boxes from specific segments to specific tile ports

Continuing from Issue #220 -- I'm trying to create an architecture configuration for the ICE40 based on [Clifford's IceStorm](http://www.clifford.at/icestorm/). I've successfully described the tile layout and the logic blocks. However, I'm having trouble figuring out how to describe interconnect / routing.

#### Expected Behaviour

There are a couple of different interconnect inside the ICE40. The segments are called Span 4 and Span 12. Each segment can only be connected to specific other segments and specific ports on a logic tile.

#### Current Behaviour

Looking at https://docs.verilogtorouting.org/en/latest/arch/reference/#switches there doesn't seem to be a way to specify the explict connections? Looking at https://docs.verilogtorouting.org/en/latest/arch/reference/#custom-switch-blocks there seems like there could be a way to do the configuration.

Looking at https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-<connection_blockinput_switch_name=

#### Possible Solution

Some way to specify the explicit connections?

#### Context

As mentioned, I'm trying to get vpr to do place and route for the ICE40.

mithro, 2017-11-10:
I did just find the following at https://docs.verilogtorouting.org/en/latest/vpr/debug_aids/

> Finally, if you need to route an FPGA whose routing architecture cannot be described in VPR’s architecture description file, don’t despair! The router, graphics, sanity checker, and statistics routines all work only with a graph that defines all the available routing resources in the FPGA and the permissible connections between them. If you change the routines that build this graph (in rr_graph*.c) so that they create a graph describing your FPGA, you should be able to route your FPGA. If you want to read a text file describing the entire routing resource graph, call the dump_rr_graph subroutine.

Is this the only way to create the type of routing that I'm looking for?

mithro, 2017-11-10:
There seems to be a `--read_rr_graph` file -- not quite sure what it is used for?
```
  --read_rr_graph RR_GRAPH_FILE                                         
                    The routing resource graph file to load. The loaded routing                          
                    resource graph overrides any routing architecture specified                          
                    in the architecture file.
  --write_rr_graph RR_GRAPH_FILE
                    Writes the routing resource graph to the specified file
  --dump_rr_structs_file DUMP_RR_STRUCTS_FILE
                    Dumps RR graph related structures to a file
```

kmurray, 2017-11-10:
**Overview**

VPR's routing stage operates on the abstract Routing Resource Graph (RR Graph) which describes all the routing wires (graph nodes) and switches (graph edges) between them.

VPR supports two methods of creating the RR Graph:

1. VPR can construct an RR Graph from a high level specification provided in the [architecture file](https://docs.verilogtorouting.org/en/latest/arch/reference/).
In this case the user specifies at a high level what routing resource are available (wire segments, switches) and how they are interconnected. Since this is a high level (incomplete) specification VPR will then try to algorithmicly construct a 'good' RR graph that meets the specification.
This has been the method historically used in VPR (given it's academic origins), since it allows easy exploration of different FPGA routing architectures. Now, while this approach can likely capture the broad characteristics of the routing architecture you are interested in (e.g. which wires/switches/block ports interconnect, the aggregate numbers of those connections), it is not really aimed at reproducing an existing architecture _exactly_.

2. VPR can load (`--read_rr_graph`) a [text description of the RR graph](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-file-format-xml). 
This is a full specification of the RR graph, and should allow _any_ RR graph to be specified - including those VPR doesn't have support for building with (1).
(You can also write out an RR graph VPR constructed with method (1) using the `--write_rr_graph` option; likely a good way to see what the format should look like.)

**Recommended Approach**

Ultimately if you want to be able to route _precisely_ the architecture of existing ICE40 devices you will need to use method (2). (While you should be able to describe a routing architecture which is very similar to the ICE40 devices with method (1), it is very unlikely the resulting architecture would be _exactly_ the same.)

For method (2), I'd recommend writing a tool which will translate IceStorm's ChipDB format into an RR Graph file which can then be loaded into VPR. Combined with an architecture file which described the blocks and device floorplan this should be enough for VPR to pack/place/route designs onto the architecture.

To generate a final bitstream, you would need to write another tool to process the .net/.place/.route files produced by VPR into a format IceStorm's tools can use to make the final bitstream.

kmurray, 2017-11-10:
**Specifics to method (1)**

While method (2) is the approach you want to take if you want to produce bitstreams for specific devices, you can model most of the routing architecture features using method (1):

> The segments are called Span 4 and Span 12.

You can specify the type of wire segments and their relative frequency with the [`<segment>`](https://docs.verilogtorouting.org/en/latest/arch/reference/#wire-segments) tag.

> Each segment can only be connected to specific other segments

Connectivity between wire segments is usually termed the 'switch block' in FPGA architecture. 

You can specify [custom switchblocks](https://docs.verilogtorouting.org/en/latest/arch/reference/#custom-switch-blocks) in the architecture file, which will describe the connectivity between wires (including which wires connect to each other, and what the pattern is).

> Each segment can only be connected to [...] specific ports on a logic tile.

Connectivity between wire segments and block is usually termed the 'connection block' in FPGA architecture.

You can specify connectivity between segments/wires and ports using the `<fc>` and [`<fc_override>`](https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-%3Cfc_overridefc_type=) tags.

kmurray, 2017-11-10:
If you encounter other issues/have questions let us know.

mithro, 2017-11-10:
Thanks! That is the conclusion I had come to from finding the documentation here -> https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-file-format-xml

Sorry about this barrage of questions, I'll try and submit a pull request which improves the documentation with your answers!

 * What is a channel? Is it just a group of wires that are grouped together? 
     * What are the requirements for a channel? I **think** that I should be using a channel for the 8 * Span 4 wires which go from tile XY to tiles (X+5)Y stopping in tiles (X+1)Y, (X+2)Y, (X+3)Y, (X+4)Y. However, each wire can be connected to a different set of other wires & IO ports in each tile. And a `<node>` seems to reference a channel rather than a wire?
     * Does grouping together these wires into a channel help or hinder anything?
     * How do you specify multiple channels? The example @ https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-format-example seems to only have a single channel and the fact x_list / y_list stuff is at the same level as the `<channel>` tag makes things a little confusing.

 * I don't quite understand `<block_types>` objects? Do these correspond to the `<pb_type>` objects in the FPGA architecture description doc?

 * What does vpr do it you don't specify timing or sizing information? Do I need to come up with dummy values or can I just leave them out initially?

 * Without actually doing any place-n-route does vpr have;
    * A way to check XML architecture / rr_graph files? (BTW Is there a schema of the XML somewhere?)
    * A way of visualizing the XML architecture / rr_graph files? (I noticed you have some nice pictures in your documentation, are these auto-generated or hand generated?)

 * Does vpr require a CLA or DCO for contributing?

kmurray, 2017-11-10:
> I'll try and submit a pull request which improves the documentation with your answers!

That would be great, and much appreciated! 

The documentation could definitely benefit from improvement.

> What is a channel?

VPR assumes that the routing wires are logically located 'adjacent' to the blocks in the FPGA floorplan. Grouped together these form vertical or horizontal 'channels' between the blocks. VPR assumes these channels span the whole dimension of the device; For example, the vertical channel with x-coordinate 0 is the first vertical channel from the left, and runs from the bottom to top of the device. See [here](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#fig-fpga-coord-system) for a diagram of the co-ordinate system. 

VPR requires (largely for historical reasons) some information about what the channel widths (number of routing wires) in each channel.

> a `<node>` seems to reference a channel rather than a wire?

A node generally represents a 'conductor' typically a wire (type CHANX for horizontal, CHANY for vertical), a block pin (IPIN/OPIN), or a source/sink (SOURCE/SINK).

An edge represents a programmable [switch](https://docs.verilogtorouting.org/en/latest/arch/reference/#switches) between two conductors.

> Does grouping together these wires into a channel help or hinder anything?

It is largely a way to logically structure the routing. It is used by the drawing code to draw the RR graph in interactive graphics. 

It has more significance internally if VPR is constructing the RR graph from a specification with method (1) above. 

The router works only with the RR graph so doesn't really have any conception of the channels.

> How do you specify multiple channels?

Each x_list/y_list represents an individual channel (location specified by the index attribute).
I think the `<channel>` tag is trying to capture statistics over all x_list/y_lists.

This is certainly an area where the file format could be improved

> I don't quite understand <block_types> objects? Do these correspond to the <pb_type> objects in the FPGA architecture description doc?

Each `<block_type>` corresponds to a top-level `<pb_type>` in the `<complexblocklist>` section of the architecture. `<block_type>`'s describe the 'interface' between the block and the RR graph.

> What does vpr do it you don't specify timing or sizing information? Do I need to come up with dummy values or can I just leave them out initially?

I believe the timing information is optional (and is just zeroed out if not specified).

Sizing info is relevant for calculating statistics about the FPGA architecture, and probably not relevant in your case (although the parser may expect it).

> A way to check XML architecture / rr_graph files? (BTW Is there a schema of the XML somewhere?)

There is no schema for either the architecture or rr_graph files. The implementations in libarchfpga/VPR are the effective specifications.

The best approach I can think of is to take a simple circuit (e.g. a single AND gate) and run it through VPR with your architecture file and rr_graph.

We've only ever tested the --read_rr_graph with RR graph's produced by VPR's --write_rr_graph, so you may encounter issues.

Note that the code does try to do error checking and give useful error messages, but this is incomplete.
If you encounter issues please file issues/patches.

> A way of visualizing the XML architecture / rr_graph files? (I noticed you have some nice pictures in your documentation, are these auto-generated or hand generated?)

You can run VPR with the [--disp on](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-disp) option, which will turn on the interactive graphics. During placement this will show the device grid, and during routing you can visualize the RR graph with the 'RR Graph' button.

> Does vpr require a CLA or DCO for contributing?

Currently we do not have any CLA in place (but this is something we should probably consider).

mithro, 2018-02-12:
I think we can close this issue as it has been superseded by #286.

closed 2018-02-12 by mithro