verilog-to-routing/vtr-verilog-to-routing #243: how to change cost in routing

sinavtr, 2017-11-16:
Hello
I have one question about VPR codes.
I gonna to change the cost of the edge(some edge in the graph) in the routing part.  in other words, I want to change the nets routing in the same placement of CLBs. for example, first, we place and rout one benchmark with seed 1. now we want to place and rout the same benchmark with seed one but we want this routing is different from the previous routing.  I guess, for this work we should change pathfinder_update_one_cost function in the routing_command file. is it true?

kmurray, 2017-11-17:
There are a couple of different approaches you could take:

1. Modifying Pathfinder Cost Parameters (easiest)
There are [command-line parameters](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#router-options) (e.g. `--pres_fac_mult`, `--acc_fac`) which control how the pathfinder algorithm calculates costs. Note that changing these parameters may significantly change the router run-time and result quality.

2. Modify the RR Node Base Costs (more work)
You can modify the basic 'costs' of RR nodes. VPR stores these in `t_rr_index_data` structs, which are initialized with the `alloc_and_load_rr_indexed_data()` function in `vpr/src/route/rr_graph_indexed_data.cpp`. Changing these values by small amounts will likely cause small changes in the router run-time and result quality.

(1) is much easier but may cause large result quality changes. (2) is more work, but would likely change the result quality less (for small changes, more akin to a seed change during placement).

I'd recommend starting with (1), and then trying (2) only if you need to.

sinavtr, 2017-11-17:
thank you very much and wish best for you.

closed 2017-11-17 by kmurray