verilog-to-routing/vtr-verilog-to-routing #246: Invalid RR constructed when pass_gate switches used

kmurray, 2017-11-21:
As noted in #245 VPR's RR Graph generator produces an invalid RR graph when `pass_gate` type switches are used.

#### Expected Behaviour

VPR should construct a legal RR graph with `pass_gate` type switches are used.

#### Current Behaviour

VPR fails during `check_rr_graph()`.
```
Error 1: 
Type: Routing
File: /project/trees/vtr/vpr/src/route/check_rr_graph.cpp
Line: 527
Message: in check_pass_transistors:
connection from node 142666 to node 144431 uses a pass transistor (switch type 0)
but there is no corresponding pass transistor edge in the other direction.
```
#### Possible Solution

We should either fix support for `pass_gate` switches, or remove support for them.
Most modern FPGA architectures using buffered switches (`tristate` for bi-dir, `mux` for uni-dir).

#### Steps to Reproduce (for bugs)

1. Run VPR with the [example architecture](https://gist.github.com/mithro/4def2abc8d20841d872ea3ca0dbf775c#file-ice40-virt-xml) with a simple BLIF nelist (e.g. `ex5p`).

#### Context

Currently we do not test any architectures using `pass_gate` switches. If we end-up fixing support for them, we should make sure to add regression tests for them to catch future bugs.

#### Your Environment

* VTR revision used: 8.0.0-dev+b10e710
* Operating System and version: Ubuntu 16.04
* Compiler version: gcc-5.4

mithro, 2018-04-27:
@kmurray Just came across this bug again. Here is a much simpler architecture which demonstrates the issue;
```xml
<?xml version="1.0"?>
<!-- set: ai sw=1 ts=1 sta et -->
<architecture xmlns:xi="http://www.w3.org/2001/XInclude">
  <models/>
  <layout>
    <fixed_layout height="3" name="1x1" width="7">
      <!--
      0123456
    2 .......
    1 .I.X.O.
    0 .......
    -->
      <col priority="10" startx="0" type="EMPTY"/>
      <col priority="10" startx="1" type="BLK_IG-IBUF"/>
      <col priority="10" startx="2" type="EMPTY"/>
      <col priority="10" startx="3" type="BLK_TI-TILE"/>
      <col priority="10" startx="4" type="EMPTY"/>
      <col priority="10" startx="5" type="BLK_IG-OBUF"/>
      <col priority="10" startx="6" type="EMPTY"/>
      <row priority="11" starty="2" type="EMPTY"/>
      <!--               starty="1"                   -->
      <row priority="11" starty="0" type="EMPTY"/>
    </fixed_layout>
  </layout>
  <complexblocklist>
    <!-- An IO pin found on an FPGA -->
    <pb_type capacity="1" name="BLK_IG-IBUF">
      <output name="I" num_pins="1"/>
      <pb_type blif_model=".input" name="BLK_BB-IBUF" num_pb="1">
        <output name="inpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="BLK_BB-IBUF.inpad" name="IBUF" output="BLK_IG-IBUF.I"/>
      </interconnect>
      <fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
      <pinlocations pattern="custom">
        <loc side="right">BLK_IG-IBUF.I</loc>
      </pinlocations>
    </pb_type>
    <!-- An IO pin found on an FPGA -->
    <pb_type capacity="1" name="BLK_IG-OBUF">
      <input name="O" num_pins="1"/>
      <pb_type blif_model=".output" name="BLK_BB-OBUF" num_pb="1">
        <input name="outpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="BLK_IG-OBUF.O" name="OBUF" output="BLK_BB-OBUF.outpad"/>
      </interconnect>
      <fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
      <pinlocations pattern="custom">
        <loc side="right">BLK_IG-OBUF.O</loc>
      </pinlocations>
    </pb_type>
    <pb_type name="BLK_TI-TILE">
      <!-- Tile Inputs -->
      <clock name="CLK" num_pins="2"/>
      <input name="IN" num_pins="6"/>
      <!-- Tile Outputs -->
      <output name="OUT" num_pins="2"/>
      <!-- Internal LUTFF -->
      <pb_type name="BLK_SI-LUTFF" num_pb="1">
        <input equivalent="false" name="I" num_pins="4"/>
        <clock equivalent="false" name="C" num_pins="1"/>
        <output equivalent="false" name="O" num_pins="1"/>
        <pb_type name="BLK_IG-LUTFF" num_pb="1">
          <clock equivalent="false" name="C" num_pins="1"/>
          <input equivalent="false" name="I" num_pins="4"/>
          <output equivalent="false" name="O" num_pins="1"/>
[... 95 lines cut ...]
```

The error with this `arch.xml` file you get the following error message;
```
Error 1: 
Type: Routing
File: /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/route/check_rr_graph.cpp
Line: 120
Message: in check_rr_graph: node 24 connects to node 113 2 times but edges are not a buffer and pass transisitor.

Makefile:91: recipe for target 'wire.disp' failed
make: *** [wire.disp] Error 1
```

Currently using;
```
VPR FPGA Placement and Routing.
Version: 8.0.0-dev+vpr-7.0.5-5790-g3c7ea4b52
Revision: vpr-7.0.5-5790-g3c7ea4b52
Compiled: 2018-04-26T19:53:34 (release build)
Compiler: GNU 6.3.0 on Linux-4.9.0-5-amd64 x86_64
University of Toronto
vtr-users@googlegroups.com
This is free open source code under MIT license.
```

kmurray, 2018-05-26:
Should be fixed as of 9e62d93.

There were two main issues:

1. The failing error check considered (duplicate edges between two nodes) only had an exception for tristate buffers and pass gates. This has been generalized to only error if there are duplicate edges of the same switch.

2. The RR graph generator produced RR graphs with multiple duplicate edges. This has been resolved tracking the edges created and only creating a new edge if an equivalent edge doesn't already exist.

closed 2018-05-26 by kmurray