verilog-to-routing/vtr-verilog-to-routing #247: Loading rr_graph.xml is broken

mithro, 2017-11-25:
I'm running the following;
```bash
	$(VPR) $(ARCH_FILE) ff.blif --disp on --write_rr_graph rr_graph.xml --route_chan_width 6
	$(VPR) $(ARCH_FILE) ff.blif --disp on --read_rr_graph rr_graph.xml --route_chan_width 6
```

The problem is best illustrated by these images;

###### Before
[image]

###### After
[image]

```patch
diff --git a/vpr/src/draw/draw.cpp b/vpr/src/draw/draw.cpp
index b4e3fa1e0..58d62f6bb 100644
--- a/vpr/src/draw/draw.cpp
+++ b/vpr/src/draw/draw.cpp
@@ -1461,12 +1461,12 @@ static void draw_chanx_to_chanx_edge(int from_node, int to_node,
                if (device_ctx.rr_nodes[to_node].direction() != BI_DIRECTION) {
                        /* must connect to to_node's wire beginning at x2 */
                        if (to_track % 2 == 0) { /* INC wire starts at leftmost edge */
-                               VTR_ASSERT(from_xlow < to_xlow);
+                               //VTR_ASSERT(from_xlow < to_xlow);^M
                                x2 = to_chan.left();
                                /* since no U-turns from_track must be INC as well */
                                x1 = draw_coords->tile_x[to_xlow - 1] + draw_coords->get_tile_width();
                        } else { /* DEC wire starts at rightmost edge */
-                               VTR_ASSERT(from_xhigh > to_xhigh);
+                               //VTR_ASSERT(from_xhigh > to_xhigh);^M
                                x2 = to_chan.right();
                                x1 = draw_coords->tile_x[to_xhigh + 1];
                        }
```

```
Build routing resource graph took 0.002002 seconds  
Build rr_graph took 0.003247 seconds.               
Confirming router algorithm: TIMING_DRIVEN.         
---- ------ ------- ------------------- ----------------- -------- ---------- ---------- ---------- ---------- -------- ------------------                                                                         
Iter   Time  Re-Rtd   Overused RR Nodes        Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ    Est Overused RR                                                                         
      (sec)   Conns                                           (ns)       (ns)       (ns)       (ns)       (ns)     Iter              Slope                                                                         
---- ------ ------- ------------------- ----------------- -------- ---------- ---------- ---------- ---------- -------- ------------------                                                                         
   1    0.0       2        1 ( 0.0598%)         6 ( 2.9%)    0.411    -0.7759     -0.411          0      0.000      N/A                N/A                                                                         
/usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/draw/draw.cpp:2593 draw_pin_to_chan_edge: Assertion 'grid_type->pinloc[grid_tile.width_offset][grid_tile.height_offset][pin_rr.side()][pin_rr.pin_num()]' failed (Pin coordinates should match block type pin locations).  
```

When we get past the asserts, the pins also appear to be drawn on the wrong edge of the block too.

I'm guessing someone has an `x, y` coordinates swapped somewhere in the reading.

The architecture file I'm using can be found here -> https://gist.github.com/mithro/ee8ee4424ecf70ffbe4bf231c21f3cf3

#### Your Environment

* VTR revision used: 43b74315f5d2abbce35ad42bb988ed7e6c708e04

```
VPR FPGA Placement and Routing.
Version: 8.0.0-dev+c658587f8-dirty
Revision: c658587f8-dirty
Compiled: 2017-11-25T13:26:04 (release build)
```
* Operating System and version: Linux

mithro, 2017-11-26:
So, I added a patch to use the `--echo_file on` flags to produce an `rr_graph.echo` file after loading (see https://github.com/mithro/vtr-verilog-to-routing/commit/6859f98024984c9b6dd33d2393d46540ba230f01 ) and I get a diff between the echo runs which looks kind of like this;
```patch
--- temp1/rr_graph.echo 2017-11-25 18:13:04.759282399 -0800
+++ temp2/rr_graph.echo 2017-11-25 18:13:04.883280713 -0800
@@ -19,21 +19,21 @@
 R: 0  C: 0
 Cost_index: 3
 
-Node: 3 IPIN (0, 1) Ptc_num: 0 Length: 0 Side: RIGHT 
+Node: 3 IPIN (0, 1) Ptc_num: 0 Length: 0 Side: TOP 
 1 edge(s): 0
 Switch types: 2
 Capacity: 1
 R: 0  C: 0
 Cost_index: 3
 
-Node: 4 IPIN (0, 1) Ptc_num: 0 Length: 0 Side: BOTTOM 
+Node: 4 IPIN (0, 1) Ptc_num: 0 Length: 0 Side: TOP 
 1 edge(s): 0
 Switch types: 2
 Capacity: 1
 R: 0  C: 0
 Cost_index: 3
 
-Node: 5 IPIN (0, 1) Ptc_num: 0 Length: 0 Side: LEFT 
+Node: 5 IPIN (0, 1) Ptc_num: 0 Length: 0 Side: TOP 
 1 edge(s): 0
 Switch types: 2
 Capacity: 1
@@ -47,21 +47,21 @@
 R: 0  C: 0
 Cost_index: 2
 
-Node: 7 OPIN (0, 1) Ptc_num: 1 Length: 0 Side: RIGHT 
+Node: 7 OPIN (0, 1) Ptc_num: 1 Length: 0 Side: TOP 
 2 edge(s): 1609 1608
 Switch types: 1 1
 Capacity: 1
 R: 0  C: 0
 Cost_index: 2
 
-Node: 8 OPIN (0, 1) Ptc_num: 1 Length: 0 Side: BOTTOM 
+Node: 8 OPIN (0, 1) Ptc_num: 1 Length: 0 Side: TOP 
 0 edge(s):
 Switch types:
 Capacity: 1
 R: 0  C: 0
 Cost_index: 2
 
-Node: 9 OPIN (0, 1) Ptc_num: 1 Length: 0 Side: LEFT 
+Node: 9 OPIN (0, 1) Ptc_num: 1 Length: 0 Side: TOP 
 0 edge(s):
 Switch types:
 Capacity: 1
@@ -89,21 +89,21 @@
 R: 0  C: 0
 Cost_index: 3
 
-Node: 13 IPIN (0, 2) Ptc_num: 0 Length: 0 Side: RIGHT 
+Node: 13 IPIN (0, 2) Ptc_num: 0 Length: 0 Side: TOP 
 1 edge(s): 10
 Switch types: 2
 Capacity: 1
 R: 0  C: 0
 Cost_index: 3
```
```patch
--- temp1/rr_graph.echo 2017-11-25 18:13:04.759282399 -0800
+++ temp2/rr_graph.echo 2017-11-25 18:13:04.883280713 -0800
...
-Node: 1575 CHANX (4, 2) Ptc_num: 4 Length: 0 Direction: BI_DIRECTION 
+Node: 1575 CHANX (4, 2) Ptc_num: 4 Length: 0 Direction: INC_DIRECTION 
 5 edge(s): 1573 1666 1648 1661 1645
 Switch types: 1 0 0 0 0
 Capacity: 1
 R: 101  C: 4.004e-14
 Cost_index: 6
 
-Node: 1576 CHANX (4, 2) Ptc_num: 5 Length: 0 Direction: BI_DIRECTION 
+Node: 1576 CHANX (4, 2) Ptc_num: 5 Length: 0 Direction: INC_DIRECTION 
 5 edge(s): 1574 1659 1653 1664 1652
 Switch types: 1 0 1 1 1
 Capacity: 1
 R: 101  C: 4.004e-14
 Cost_index: 6
 
-Node: 1577 CHANX (1, 3) Ptc_num: 0 Length: 0 Direction: BI_DIRECTION 
+Node: 1577 CHANX (1, 3) Ptc_num: 0 Length: 0 Direction: INC_DIRECTION 
 5 edge(s): 1577 1618 1617 1629 1613
 Switch types: 0 0 1 1 1
 Capacity: 1
 R: 101  C: 8.004e-14
 Cost_index: 4
 
-Node: 1578 CHANX (1, 3) to (2, 3) Ptc_num: 1 Length: 1 Direction: BI_DIRECTION 
+Node: 1578 CHANX (1, 3) to (2, 3) Ptc_num: 1 Length: 1 Direction: INC_DIRECTION 
 8 edge(s): 1578 1644 1631 1604 1632 1618 1627 1607
 Switch types: 0 1 1 0 0 0 0 0
 Capacity: 1
 R: 202  C: 9.531e-14
 Cost_index: 4
 
-Node: 1579 CHANX (1, 3) to (3, 3) Ptc_num: 2 Length: 2 Direction: BI_DIRECTION 
+Node: 1579 CHANX (1, 3) to (3, 3) Ptc_num: 2 Length: 2 Direction: INC_DIRECTION 
 11 edge(s): 1587 1656 1643 1630 1605 1646 1633 1634 1619 1624 1612
 Switch types: 0 1 1 1 0 0 0 0 0 0 0
 Capacity: 1
 R: 303  C: 1.0658e-13
 Cost_index: 5
 
-Node: 1580 CHANX (1, 3) to (4, 3) Ptc_num: 3 Length: 3 Direction: BI_DIRECTION 
+Node: 1580 CHANX (1, 3) to (4, 3) Ptc_num: 3 Length: 3 Direction: INC_DIRECTION 
 14 edge(s): 1669 1648 1655 1638 1627 1612 1666 1647 1646 1634 1633 1624 1619 1605
 Switch types: 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 Capacity: 1
 R: 404  C: 1.4185e-13
 Cost_index: 5
 
...
```

mithro, 2017-11-26:
Fixed in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/251

kmurray, 2017-11-27:
This seems to have only effected drawing (the actual graph nodes and edges used by routing seem to have been loaded correctly).

We have regression test that verify VPR produces the same routing result when loading the same RR graph it generated -- but they don't check that the graph is drawn the same.

Thanks again for the fix!

closed 2017-11-27 by kmurray

mithro, 2017-11-27:
@kmurray Could you point me to the regression tests? I would like to add one which checks the `rr_graph.echo` file was the same as that would have caught this bug.

kmurray, 2017-11-27:
I was thinking the same thing, hence abe90c3.

We now do a text diff of the 1st RR graph file (generated from the architecture file), and 2nd RR graph file (generated when we re-load the 1st RR graph file).

PR #251 (2017-11-27): Fix rr_graph_reader loading of direction and side attributes.
Previously newType was only ever set to NUM_RR_TYPES which meant that
the side and direction processing was never called.

Fixes #247.
commit 0234354c (2017-11-27): Fix rr_graph_reader loading of direction and side attributes.

Previously newType was only ever set to NUM_RR_TYPES which meant that
the side and direction processing was never called.

Fixes #247.