verilog-to-routing/vtr-verilog-to-routing #256: Support mode advanced flip-flop configurations by default

mithro, 2017-11-28:
Currently the only primitive flip flop supported in vpr is a simple flip-flop. However, most FPGA architectures seem to use a flip-flops which have both a "clock enable" and "set/reset" pins. Most flip flops can also work in sync or async reset configurations.

Example from iCE40 architecture;
[image]

Examples from Artix-7 architecture;
[image]

#### Expected Behaviour

Built in support for flip-flops with clock enable and set/reset functionality.

#### Current Behaviour

Currently the only primitive flip flop supported in vpr is a simple flip-flop describe by the following;
```verilog
//D-FlipFlop module
module DFF #(
    parameter INITIAL_VALUE=1'b0    
) (
    input clock,
    input D,
    output reg Q
);

    specify
        (clock => Q) = "";
        $setup(D, posedge clock, "");
        $hold(posedge clock, D, "");
    endspecify

    initial begin
        Q <= INITIAL_VALUE;
    end

    always@(posedge clock) begin
        Q <= D;
    end
endmodule
```

#### Possible Solution

Currently the work around is to define your own model with something like the following;
```xml
    <pb_type name="SB_FF" num_pb="1" blif_model=".subckt SB_FF">
     <clock  name="C" num_pins="1"/>
     <input  name="E" num_pins="1"/>
     <input  name="S" num_pins="1"/>
     <input  name="D" num_pins="1"/>
     <output name="Q" num_pins="1"/>
     <T_setup    value="10e-12" port="SB_FF.D" clock="C"/>
     <T_setup    value="10e-12" port="SB_FF.S" clock="C"/>
     <T_clock_to_Q max="10e-12" port="SB_FF.Q" clock="C"/>
  </pb_type>
```

mithro, 2017-11-28:
I assume the current support for only `BLIF .latch (DFF) primitive` doesn't support this type of flip-flop?

kmurray, 2017-11-28:
Yes, we are effectively limited by BLIF, which only supports a very simple `.latch` without more advanced features.

Defining your own FF as a `.subckt` is the correct way to model a more advanced features.

I don't think we will add support for a more advanced default FF, since the exact capabilities vary from architecture to architecture -- allowing a user defined primitive should be sufficient to handle all these cases.

closed 2017-11-28 by kmurray